olikraus/u8g2 #1671: Make I2C multi display (and thread) safe

sgjava, 2021-11-07:
Maybe thsi is a more general question about thread safety, but since I2C must maintain buffer and index https://github.com/olikraus/u8g2/blob/a8e72d8c651d653ad97d00f7a594fae59f3f1fc4/sys/arm-linux/port/u8g2port.c#L265 if another display were to sneak in before U8X8_MSG_BYTE_END_TRANSFER it could corrupt the buffer.

SPI on the other hand can be driven by U8X8_MSG_BYTE_SEND only thus no function level statics are required to persist across messages. Forgive if I'm misunderstanding something, but I tried the same thing with I2C (send buffer when U8X8_MSG_BYTE_SEND called) and it didn't work.

I realize this might only be an issue for Linux ports.

olikraus, 2021-11-07:
Indeed the second approach provided here: https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform#hardware-i2c-communication is not thread safe.

It is not an issue with Arduino, because the buffer is part of the wire lib instead. It is also not an issue with embedded systems (because usually only static scheduling will be used), but yes, it might be an issue on linux if multiple displays are used.

Probably START_TRANSFER should lock a mutex (https://pubs.opengroup.org/onlinepubs/7908799/xsh/pthread_mutex_init.html) , which is then checked in BYTE_SEND and START_TRANSFER and released in END_TRANSFER

sgjava, 2021-11-07:
Hmmm, could we do an ifdef in u8x8_struct with a buffer pointer and index? Then that would be at display level. It wouldn't impact and of the micro controllers that are doing things in a synchronized fashion since it wouldn't be used. I think that's the last piece missing to allow multiple I2C displays in a thread safe fashion.

olikraus, 2021-11-07:
To some extend I agree, that there should be a device specific aux user pointer, which could then point to a custom struct with the buffer. In fact you could use the already existing user pointer: https://github.com/olikraus/u8g2/blob/master/csrc/u8x8.h#L372

sgjava, 2021-11-07:
Hmmnm, what about index. I assume that can change before sending buffet? I mean I could track it in the port via bus number like I'm doing GPIO pins.

sgjava, 2021-11-07:
You know what that's a great idea. I'll just use a strict!

closed 2021-11-07 by sgjava

sgjava, 2021-11-08:
@olikraus , so you have me thinking. I moved the buffer to a struct using user_ptr, but why not move all the bus info there as well? My original idea was taken from the PR with the I2C bus in the u8x8_struct, but since most of this lives in the port so to speak, should I refactor all that info into my new struct and manage in the port? I wish I caught wind of user_ptr sooner :). Not a big deal, but where do you think all that info should live? Inside u8x8_struct or user_ptr via the arm-linux port? Seems like less ifdef complexity???

olikraus, 2021-11-08:
Putting everything into a separate struct seems to be a more cleaner method to me.

sgjava, 2021-11-08:
@olikraus I agree, let me refactor to make that a reality. It also makes sense to move the GPIO inside the user data struct since I could have multiple DC and RESET pins on different buses. This will make everything cleaner and more flexible. As a side effect is I2C now becomes thread safe.