olikraus/u8g2 #1674: SSD1327 u8g2.drawXBM does not write to top 8 pixels

trevor8feb, 2021-11-10:
Hello,

I have two 1.5" OLED displays. 
One is driven using u8g2 to write a bitmap starting at the top left of the screen (0,0).
The other uses the Waveshare example sketch OLED_Show to write bitmaps to the top 8 pixels.
I can't understand why u8g2 does not write to the top row of 8 pixels.
Photo attached.

In all other respects the u8g2 works very well - thanks.
Any help would be appreciated.

The bitmap:
#define VU_20scale7_width 128
#define VU_20scale7_height 64
static unsigned char VU_20scale7_bits[] = {
   0x7f, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x05, 0x00, 0x00, 0x24, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x05, 0x02, 0x00, 0x14, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xc0, 0x1d, 0x02, 0x00,
   0x24, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x80,
   0x27, 0xa5, 0x0f, 0x00, 0x44, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0xc0, 0x1d, 0x02, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x02, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xf0, 0x00, 0xf0, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xe0, 0x01, 0x10,
   0x00, 0x80, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x20, 0x01, 0xf0, 0x00, 0xe0, 0x00, 0x10, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x90, 0x03, 0x00, 0x01, 0x80, 0x00, 0x80, 0x00, 0x18,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x58, 0x04, 0x00, 0x01, 0xf0,
   0x00, 0xf0, 0x00, 0x10, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x50,
   0x04, 0x00, 0x01, 0x00, 0x00, 0x00, 0x00, 0x10, 0x80, 0x03, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x50, 0x04, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x10,
   0x40, 0x04, 0x01, 0x00, 0x00, 0x00, 0x00, 0x90, 0x03, 0x00, 0x00, 0x20,
   0x20, 0x20, 0x00, 0x10, 0x40, 0x84, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x02, 0x20, 0x20, 0x20, 0x20, 0x00, 0x40, 0x04, 0x01, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x02, 0x20, 0xe0, 0x2f, 0x20, 0x20,
   0x80, 0x03, 0x81, 0x07, 0x00, 0x00, 0x00, 0x00, 0x01, 0x00, 0x82, 0xff,
   0x3f, 0xf8, 0x23, 0x20, 0x20, 0x00, 0x01, 0x04, 0xf0, 0x1c, 0x00, 0x00,
   0x03, 0xfc, 0xff, 0x02, 0x00, 0x00, 0xfe, 0x6f, 0x20, 0x00, 0x81, 0x07,
   0x80, 0x22, 0x00, 0x00, 0xfa, 0x07, 0x00, 0x02, 0x00, 0x00, 0x00, 0xf8,
   0x31, 0x60, 0x80, 0x00, 0xf0, 0x22, 0x00, 0x00, 0x07, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0xff, 0x33, 0x80, 0x07, 0x10, 0x22, 0x08, 0xc0,
   0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xf0, 0x1f, 0x40, 0x00,
   0xf0, 0x1c, 0x18, 0x7c, 0x00, 0x00, 0xe0, 0x39, 0x00, 0x00, 0x00, 0x00,
   0x18, 0xfc, 0x61, 0x00, 0x00, 0x00, 0xf0, 0x07, 0x00, 0x00, 0x20, 0x44,
   0x00, 0x00, 0x00, 0x00, 0x08, 0xe0, 0x3f, 0x00, 0x00, 0x10, 0x30, 0x00,
   0x00, 0x00, 0xe0, 0x45, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x1f, 0x00,
   0x00, 0x30, 0x1e, 0x00, 0x00, 0x00, 0x00, 0x45, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0xf0, 0x21, 0x00, 0xe0, 0x01, 0x00, 0x00, 0x00, 0xe0, 0x39,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xe0, 0x37, 0x00, 0x78, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x0c, 0x00, 0x1f,
   0x00, 0xcf, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x20,
   0x63, 0x92, 0x00, 0x0c, 0xc0, 0x81, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0xb0, 0x94, 0xd2, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xa0, 0x94, 0x4c, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xa0,
   0x94, 0x20, 0x03, 0x00, 0x00, 0x40, 0x0e, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x20, 0x63, 0xa0, 0x04, 0x00, 0x00, 0x60, 0x11, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x90, 0x04, 0x00,
   0x00, 0x40, 0x11, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x03, 0x00, 0x00, 0x40, 0x11, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x40, 0x0e, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x7c, 0x20, 0x04, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x30, 0xbc, 0x07,
   0x44, 0x20, 0x04, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x48, 0x20, 0x04, 0x44, 0x6e, 0x76, 0x33, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x48, 0x20, 0x04, 0x3c, 0x51, 0x8a, 0x1e,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x48, 0x30, 0x06,
   0x24, 0x4f, 0x8a, 0x0c, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0xfc, 0x10, 0x02, 0x64, 0xc1, 0x8b, 0x1c, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x84, 0x18, 0x03, 0x44, 0x91, 0x89, 0x1e,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x84, 0x08, 0x01,
   0x44, 0x8e, 0x71, 0x33, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x84, 0x08, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00 };

Part of sketch:

  u8g2.firstPage();
  do{
//  u8g2.setFont(u8g2_font_ncenB14_tr);
//  u8g2.setCursor(0, 0);
//  u8g2.print("Hello World!");
  u8g2.drawXBM( 0, 0, VU_20scale7_width, VU_20scale7_height,VU_20scale7_bits);
  u8g2.drawLine(a1, a2, hMeter, vMeter);
  } while (u8g2.nextPage());  
}
[image]

olikraus, 2021-11-10:
Hmm... more generally you are saying you can not access the top 8 pixel, right?
Which constructor do you use.
If you would use the u8g2 page buffer Flip Mode example (from the Arduino IDE examples or from here: https://github.com/olikraus/u8g2/tree/master/sys/arduino/u8g2_page_buffer/FlipMode) will you see a full frame border all the time?

In fact there are many different types of SSD1327 displays and maybe you have just selected the wrong constructor:
SSD1327 WS_96X64
SSD1327 SEEED_96X96
SSD1327 EA_W128128
SSD1327 MIDAS_128X128
SSD1327 VISIONOX_128X96

trevor8feb, 2021-11-10:
Hello,

Thanks for the quick response - you are quite correct;
I changed the constructor to U8G2_SSD1327_WS_128X128_1_4W_SW_SPI
u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 7, /*
reset=*/ 8); and it works perfectly.
The flip mode example drew the box around the whole display.
My bitmap now starts at pixel 0,0.

I can now replace the broken VU meters on my 1970's reel to reel tape
recorder with digital equivalents !

Many, many thanks

Trevor King

[image: oled_1_5_solved.jpg]

closed 2022-03-04 by olikraus