olikraus/u8g2 #1675: Software I2C performance

sgjava, 2021-11-10:
When I do I2C software it's so slow I can watch the buffer render. It seems https://github.com/olikraus/u8g2/blob/a8e72d8c651d653ad97d00f7a594fae59f3f1fc4/sys/arm-linux/port/u8g2port.c#L144 might be too slow. When I did code for capturing image frames in another project I'd calculate the remaining time (delta) and only sleep for that period instead of a fixed delay. I'm not sure that would work here, but the Duo can do 242 KHz  https://github.com/sgjava/java-periphery#gpio-performance-using-perf bit banging on a single core, so I should get close to 100 KHz in software depending on what your code is doing upstream. Your thoughts?

sgjava, 2021-11-10:
@olikraus I tested by totally removing delay from U8X8_MSG_DELAY_I2C and it's the same or a little faster than hardware I2C at 100 KHz. This is why I hate these kinds of delays. What I will do is add this to my user data struct, so on a faster board you can add a delay value. This make software I2C usable now :)

closed 2021-11-10 by sgjava

olikraus, 2021-11-10:
I did my own tests also long back where I observed that the delay is required from some displays. 
Of course I agree that the switch-case U8X8_MSG processing introduces its own delays, which might be even larger than the I2C clock period. 

Nevertheless design decisions for u8g2 had always been: Stay on the safe side. If a display doesn't work because the user had used very long wires and a bad display with slow logic on the I2C bus, then u8g2 should still work with this display.

One important aspect for U8g2 always had been: Prove that your hardware works, then use this state for further optimization (like removal of the U8X8_MSG_DELAY_I2C).

One more aspect is the addressed Arduino User Community: I feel it is better an exprienced user complains about u8g2 speed instead a inexprienced user creates issues here for a black display ;-)

sgjava, 2021-11-10:
Hahahaha, we agree about not messing things up. I put the delay value in my user data struct and if it's zero then no delay otherwise ns delay based on arg_int. This way on init you can tweak the value. It really was not your problem since this was in the arm-linux port. Now it looks like:
```
	case U8X8_MSG_DELAY_I2C:
		// arg_int is the I2C speed in 100KHz, e.g. 4 = 400 KHz
		if (arg_int != 0) {
			user_data = u8x8_GetUserPtr(u8x8);
			sleep_ns(user_data->delay * arg_int);
		}
		break;
```
I need to test SW SPI as well, but the arm-linux port becomes a lot more useful with SW driven displays. Are there any such messages for SPI delay?

sgjava, 2021-11-11:
@olikraus I'm thinking about adding a ifndef around `u8x8_gpio_Delay(u8x8, U8X8_MSG_DELAY_I2C, u8x8->display_info->i2c_bus_clock_100kHz);`, so it will compile by default, but can be removed in arm-linux. Then I will define for arm-linux.

olikraus, 2021-11-11:
@sgjava 
The slow SW I2C and SPI speed might be also caused by the GPIO control itself, which is SUPER SLOW if done in a wrong way or with wrong privileges.

I have recently used the bcm2835 library (raspi): https://github.com/olikraus/u8g2/blob/e823a8316c19fadf736289ca9e19e1a7b663fca2/sys/raspi_gpio/hal/raspi_gpio_hal.c#L62

It is super fast, but only when executed as root. It seems that the GPIO can be either manipulated via /dev/mem or via /dev/gpiomem (see details at http://www.airspayce.com/mikem/bcm2835/)
The later case seems to be very slow. 

The point is this: If you use the fast /dev/mem approach, then you need the delay, because otherwise the I2C would be too fast.

olikraus, 2021-11-11:
Maybe your gpio procedures provided by 
https://github.com/olikraus/u8g2/blob/a8e72d8c651d653ad97d00f7a594fae59f3f1fc4/sys/arm-linux/port/u8g2port.h#L9
are too slow.

sgjava, 2021-11-11:
SW I2C is pretty darn fast on NanoPi Duo and it has one of the slower bit banging speeds of the boards I have. Configuration wise I'm sure I have everything setup fine. SW I2C might even be a little faster than HW 100 KHz. I'll have to dig into SPI code and see what's up. I use a delay value, so if the GPIO is too fast you can add user based delay. Not blaming you here, just an unknown at this point.

So if you look at my project https://github.com/sgjava/java-periphery#gpio-performance-using-perf you will see where I compare GPIOD /dev/gpiochipxx (not SYSFS) and MMIO /dev/mem. GPOID is fast enough for I2C, but SPI at 500 KHz would probably require MMIO. BUt it does seem slow even for 242 KHz.

sgjava, 2021-11-11:
Oh and those routines are a lot faster than sysfs that the original author used, but not as fast as MMIO.

olikraus, 2021-11-11:
> So if you look at my project https://github.com/sgjava/java-periphery#gpio-performance-using-perf you will see where I compare GPIOD /dev/gpiochipxx (not SYSFS) and MMIO /dev/mem. GPOID is fast enough for I2C, but SPI at 500 KHz would probably require MMIO. But it does seem slow even for 242 KHz.

Nice

sgjava, 2021-11-11:
Hahaha, OK same thing as I2C, but U8X8_MSG_DELAY_NANO messages. I'll fix that too in the port.