olikraus/u8g2 #1678: NHD-C12864WC-FSW-FBW-3V3

blankenshipbotron, 2021-11-11:
Hi, I've been using a Newhaven LCD (NHD-C12864WC-FSW-FBW-3V3) and using the following constructor:
U8G2_ST7565_NHD_C12864_F_4W_HW_SPI u8g2(U8G2_R2, /* cs=*/ 9, /* dc=*/ 5, /* reset=*/ 6);

I have had to set the contrast to max 255 to get a barely legible contrast 
u8g2.setContrast(255);

I called Newhaven support and the pointed me to some Arduino sample code. I noticed that
They had a value of 0x025 for the V0 voltage resistor ratio.

Current released u8g2 code has 0x023. I changed the following to 0x025 and now it has
full range very visible at 170. At 255 its blacked out.

File name I changed u8x8_d_st7565.c

Current value
U8X8_C(0x023),		                /* set V0 voltage resistor ratio to large*/
Value I changed
U8X8_C(0x023),		                /* Im not sure wat 0x023 changes to*/

I changed line 640 of to 0x025
https://github.com/olikraus/u8g2/blob/b6faa6571f57ba0cc583add78b0909a539dacca9/csrc/u8x8_d_st7565.c#L640

I have verified on 4 displays
Thanks
mike

olikraus, 2021-11-12:
Thanks for the feedback. Still I wonder...

I will change this in the repo.

blankenshipbotron, 2021-11-12:
Hi, here is my setup for this particular LCD if it can help anyone when there setting up. I use a adafruit feather M0 (SAMD21 MCU) as the controller.
mike
[NHD-LCD-C12664WC-SETUP-HW.pdf](https://github.com/olikraus/u8g2/files/7529816/NHD-LCD-C12664WC-SETUP-HW.pdf)
e

olikraus, 2022-01-07:
Done

closed 2022-01-07 by olikraus