olikraus/u8g2 #1681: Callback return codes

sgjava, 2021-11-15:
I'd like to return error code back, but couldn't find anything in the porting documentation. For instance all the callbacks return 0 by default or 1. What happen if you send back < 0? Looking for a clean way to avoid seg faults.

olikraus, 2021-11-15:
There should be only 0 and 1 as a return value.
1: Message was handled
0: Message was not handled

There is no error code as such, however 0 could be interpreted as insuccessful attempt to handle the message.

Usually the U8x8 procedure do not check for the return value. And I also did not always use the above convention everywhere.

sgjava, 2021-11-15:
Ok, so if I had an init error I should exit I suppose then? i.e. GPIO pin in use or I2C bus doesn't exist.

olikraus, 2021-11-15:
Ahm... Well, it is not checked, u8x8/u8g2 would just continue.

sgjava, 2021-11-15:
OK, so it makes sense to bail on error, thanks.

closed 2021-11-15 by sgjava