olikraus/u8g2 #1682: New commits broke things on Linux SPI

shlyakpavel, 2021-11-15:
Previously, I could use my SPI like thishttps://github.com/shlyakpavel/u8g2/commit/3b6af74b565a1dd44734ac5513dc7ebd998281fa
Now, I need to define DC pin even if my screen does not have it. I cannot setup CS pin correctly as u8g2.initSpiHw just doesn't support setting it right. I have tried to fix these to myself, but I didn't get the screen working :(

shlyakpavel, 2021-11-15:
https://github.com/olikraus/u8g2/pull/1676 broke it

olikraus, 2021-11-15:
@sgjava ?

sgjava, 2021-11-15:
Can you give me an example of code that was working here? Then I can try locally.

shlyakpavel, 2021-11-15:
> @shlyakpavel Can you give me an example of code that was working here? Then I can try locally.

https://github.com/shlyakpavel/u8g2/blob/3b6af74b565a1dd44734ac5513dc7ebd998281fa/sys/arm-linux/examples/cpp-examples/u8g2_4wire_hw_spi/u8g2_4wire_hw_spi.cpp
This code worked well

sgjava, 2021-11-15:
I think the CS thing I just need to make it where you can pass a pin in.
`u8x8_SetPin(u8g2_GetU8x8(u8g2), U8X8_PIN_CS, U8X8_PIN_NONE);`

shlyakpavel, 2021-11-15:
There's is also a problem that I am required to set DC pin that doesn't exist on st7920

sgjava, 2021-11-15:
@shlyakpavel Yeah, I got that too. I'll make it where you can pass a < 0 value for the pin which will result in U8X8_PIN_NONE. Would that work?

shlyakpavel, 2021-11-15:
Wow
I have manually passed U8X8_PIN_NONE as OLED_SPI_PIN_DC and correct OLED_SPI_PIN_CS and everything works now. It's a dirty hack, so I won't make it a PR. Just saying this to clarify no other problems are here

shlyakpavel, 2021-11-15:
> @shlyakpavel Yeah, I got that too. I'll make it where you can pass a < 0 value for the pin which will result in U8X8_PIN_NONE. Would that work?

I don't know

shlyakpavel, 2021-11-15:
> I'll probably make changes anyways, since this may come up again with other displays.

Sure! Thank you

sgjava, 2021-11-15:
U8X8_PIN_NONE 255, so this could be a legit pin on some boards. I like the < 0 idea will set the pin to U8X8_PIN_NONE. I'll leave this open until I make PR.

sgjava, 2021-11-15:
Oh crap the args are uint8_t, let me think about this a bit. I may need to make those int for the init functions.

sgjava, 2021-11-15:
I also have to add the exception on init stuff.

olikraus, 2021-11-15:
U8X8_PIN_NONE 255: The assumtion was that there is hardly a controller in the Arduino world with such a huge number of GPIO pins.

sgjava, 2021-11-15:
@olikraus that would make my life easier. I mean if you have pin 255 you could just use another pin I guess. This would only impact hardware I2C and SPI where the pin is 255 and you have no other choice. To be honest I've not see any boards use 255 with sysfs or gpiod which would impact arm-linux port.

sgjava, 2021-11-15:
@shlyakpavel  check out https://github.com/olikraus/u8g2/pull/1684

sgjava, 2021-11-15:
This should be closed out @shlyakpavel, PR is waiting on merge from @olikraus. Changes tested out fine.

sgjava, 2021-11-15:
Thanks @olikraus