olikraus/u8g2 #1683: initialization discards 'const' qualifier from pointer target type : char *s = "999";

whogarden, 2021-11-15:
**Versions**
Arduino IDE v8.13 and Arduino core ESP32 v2.0.1
ESP32 4MB dev board 
Oled SH1106
Library : name=U8g2 / version=2.32.6
 
Compilation error in file : mui_u8g2.c line 525 / 638
**Error**
..\libraries\U8g2_Arduino-master\src\clib\mui_u8g2.c: In function 'mui_u8g2_u8_vmm_draw_wm_pi':
..\libraries\U8g2_Arduino-master\src\clib\mui_u8g2.c:525:13: warning: initialization discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
   char *s = "999";
                 ^~~~~
..\libraries\U8g2_Arduino-master\src\clib\mui_u8g2.c: In function 'mui_u8g2_u8_vmm_draw_wm_pf':
..\libraries\U8g2_Arduino-master\src\clib\mui_u8g2.c:638:13: warning: initialization discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
   char *s = "999";
                 ^~~~~

**Sample code**
```
//My dice from here https://www.instructables.com/id/Arduino-Oled-Dice/

#include<U8g2lib.h>
#include <Wire.h>
#include<SPI.h>

U8G2_SH1106_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ 16, /* clock=*/ 4, /* data=*/ 5);

const byte Roll_PIN = 0;
int dicenr1 = 2; // Dice random number 2
int dicenr2 = 4; //Dice random number 4
byte rollMode = 2;
int dicenumber = 1;

///////////////////Roll dice draw and display///////////////////
void Roll() {
  String number[25] = {"Un", "Deux", "Trois", "Quatre", "Cinq", "Six"};

  int x;
  int x1;
  const int y = 2;
  u8g2.setFont(u8g_font_helvB10);

  u8g2.firstPage();
  do {
    if (rollMode == 1) {
      u8g2.drawStr(36, 64, "Last Roll");
    } else if (rollMode == 0) {
      u8g2.drawStr(36, 64, "Roll Dice");
    }

    switch (dicenumber) {

      case (1): //Two dice
        x = 76;

        u8g2.drawFrame(2 + x, y, 48, 48);
        if (dicenr1 == 1) {
          u8g2.drawDisc(26 + x, 24 + y, 4); //Center
          x1 = 13;
        } else if (dicenr1 == 2) {
          u8g2.drawDisc(12 + x, 10 + y, 4); //upper L
          u8g2.drawDisc(40 + x, 38 + y, 4); //lower R
          x1 = 8;

        } else if (dicenr1 == 3) {
          u8g2.drawDisc(26 + x, 24 + y, 4); //Center
          u8g2.drawDisc(12 + x, 10 + y, 4); //upper L
          u8g2.drawDisc(40 + x, 38 + y, 4); //lower R
          x1 = 7;

        } else if (dicenr1 == 4) {
          u8g2.drawDisc(12 + x, 10 + y, 4); //upper L
          u8g2.drawDisc(40 + x, 10 + y, 4); //upper R
          u8g2.drawDisc(12 + x, 38 + y, 4); //lower L
          u8g2.drawDisc(40 + x, 38 + y, 4); //lower R
          x1 = 2;

        } else if (dicenr1 == 5) {
          u8g2.drawDisc(12 + x, 10 + y, 4); //upper L
          u8g2.drawDisc(40 + x, 10 + y, 4); //upper R
          u8g2.drawDisc(12 + x, 38 + y, 4); //lower L
          u8g2.drawDisc(40 + x, 38 + y, 4); //lower R
          u8g2.drawDisc(26 + x, 24 + y, 4); //Center
          x1 = 11;

        } else if (dicenr1 == 6) {
          u8g2.drawDisc(12 + x, 10 + y, 4); //upper L
          u8g2.drawDisc(40 + x, 10 + y, 4); //upper R
          u8g2.drawDisc(12 + x, 38 + y, 4); //lower L
          u8g2.drawDisc(40 + x, 38 + y, 4); //lower R
          u8g2.drawDisc(12 + x, 24 + y, 4); //center L
          u8g2.drawDisc(40 + x, 24 + y, 4); //center R
          x1 = 16;
        }
        if (rollMode == 2) {

          u8g2.setCursor(x1 + x, 62 + y);
          u8g2.print(number[dicenr1 - 1]);

        }
      case (2)://one Dice

        if (dicenumber == 2) {
          x = 40;
        } else {
          x = 5;
        }
        u8g2.drawFrame(2 + x, y, 48, 48);

        if (dicenr2 == 1) {
          u8g2.drawDisc(26 + x, 24 + y, 4); //Center
          x1 = 15;

        } else if (dicenr2 == 2) {
          u8g2.drawDisc(12 + x, 10 + y, 4); //upper L
          u8g2.drawDisc(40 + x, 38 + y, 4); //lower R
          x1 = 8;

        } else if (dicenr2 == 3) {
          u8g2.drawDisc(26 + x, 24 + y, 4); //Center
          u8g2.drawDisc(12 + x, 10 + y, 4); //upper L
          u8g2.drawDisc(40 + x, 38 + y, 4); //lower R
          x1 = 7;

        } else if (dicenr2 == 4) {
          u8g2.drawDisc(12 + x, 10 + y, 4); //upper L
          u8g2.drawDisc(40 + x, 10 + y, 4); //upper R
          u8g2.drawDisc(12 + x, 38 + y, 4); //lower L
          u8g2.drawDisc(40 + x, 38 + y, 4); //lower R
          x1 = 4;

        } else if (dicenr2 == 5) {
          u8g2.drawDisc(12 + x, 10 + y, 4); //upper L
          u8g2.drawDisc(40 + x, 10 + y, 4); //upper R
          u8g2.drawDisc(12 + x, 38 + y, 4); //lower L
          u8g2.drawDisc(40 + x, 38 + y, 4); //lower R
          u8g2.drawDisc(26 + x, 24 + y, 4); //Center
          x1 = 11;

        } else if (dicenr2 == 6) {
          u8g2.drawDisc(12 + x, 10 + y, 4); //upper L
          u8g2.drawDisc(40 + x, 10 + y, 4); //upper R
          u8g2.drawDisc(12 + x, 38 + y, 4); //lower L
          u8g2.drawDisc(40 + x, 38 + y, 4); //lower R
          u8g2.drawDisc(12 + x, 24 + y, 4); //center L
          u8g2.drawDisc(40 + x, 24 + y, 4); //center R
          x1 = 16;
        }
        if (rollMode == 2) {
          u8g2.setCursor(x1 + x, 62 + y);
          u8g2.print(number[dicenr2 - 1]);
        }
        break;
    }
  } while (u8g2.nextPage());
  delay(100);
}
/////////////////////////end of void Roll()/////////////////////////////////

void randomGenerator () {
  dicenr1 = random(1, 7);
  dicenr2 = random(1, 7);
}
//////////////////////Slow down rolling dice///////////////////////////////

void slow() {
  int d = 0;
  for (int i = 0; i < (random(7, 30)); i++) {
    randomGenerator();
    Roll();
    delay(d);
    d = d + 5;
  }
}
////////////////////////////////////////////////////////////////////////
void setup(void) {
  //oled.setI2CAddress(0x3C * 2); //set display adress if needed (default 0x3C)
  u8g2.begin();
  u8g2.setFlipMode(0);
  pinMode(Roll_PIN, INPUT_PULLUP);
  Serial.begin(115200);
  Serial.print("\nFile : My-Dice_SH1106.ino\n");
  //Start with 2 dices with drawDisc = 1
  slow();
  Roll(); 
}
void loop(){
  Serial.println (digitalRead(Roll_PIN));
  if (digitalRead(Roll_PIN) == LOW){
    slow();
    randomGenerator();
    Roll();
    delay(500);
  }
  }
  
```
**Result**
It's no more working with last Arduino core ESP32 v2.0.1
The ESP always reboot and nothing in 'ESP exception decoder'

olikraus, 2021-11-15:
BTW: The latest official release of u8g2 is 2.31.2 (https://www.arduinolibraries.info/libraries/u8g2)
2.32.6 is an untested development snapshot

whogarden, 2021-11-15:
In debug I can see : 
[ 37070][E][Wire.cpp:308] beginTransmission(): Bus is in Slave Mode
[ 37076][E][Wire.cpp:331] endTransmission(): Bus is in Slave Mode
etc

olikraus, 2021-11-15:
Hmmm I have no idea...

whogarden, 2021-11-16:
I am not an expert, 
I try to understand why I have pb only with Arduino core ESP32 v2.0.1
when I redefine pins (it's mandatory because the oled is on board and I can not change the wiring).
sda is gpio4, scl is gpio15 and reset is gpio16
[image]

I set : 
```
U8G2_SH1106_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ 16, /* clock=*/ 15, /* data=*/ 4);
```
and it looks like the sda 4 is seen as address

The second debug line is : 
```
[    10][I][esp32-hal-i2c-slave.c:234] i2cSlaveInit(): Initialising I2C Slave: sda=15 scl=22 freq=100000, addr=0x4
```
Can you tell me where I am wrong?
Thank you.

olikraus, 2022-01-07:
At least regarding the issue title: I should have fixed this now.

closed 2022-01-07 by olikraus

whogarden, 2022-01-11:
Tests OK. ;>