olikraus/u8g2 #1689: Linux hw SPI: Custom SPI mode for a display

shlyakpavel, 2021-11-16:
I need to setup custom SPI mode for a display. I have st7920 controller and I need to use mode 0 instead of 3. What I do now is a dirty hack:
[image]
Is there a way to do it correctly?

sgjava, 2021-11-16:
You mean like u8x8->display_info->spi_mode?

sgjava, 2021-11-17:
OK, so that needs to be added to user info struct. Let me take a look at that.

sgjava, 2021-11-18:
@shlyakpavel see https://github.com/olikraus/u8g2/pull/1692 I also added max_speed since I know you will ask for this next I bet. Go ahead and close @olikraus once you merge please.

olikraus, 2021-11-18:
I think the PR is already closed.

sgjava, 2021-11-18:
https://github.com/olikraus/u8g2/pull/1697, had to change name :)

olikraus, 2021-11-18:
ok, done

closed 2022-03-04 by olikraus