olikraus/u8g2 #1690: Constructor for Surenoo SLG12864I LCD Display

olikraus, 2021-11-16:
### Discussed in https://github.com/olikraus/u8g2/discussions/1688

<div type='discussions-op-text'>

<sup>Originally posted by **johngineer** November 16, 2021</sup>
I recently received a couple examples of this display: https://www.aliexpress.com/item/32913297404.html?spm=a2g0s.12269583.0.0.9f0e3b9fTgWb2X

The MPN is given as SLG12864I. It uses an ST7567 COG driver.

I didn't find it in the current list of constructors but I was able to get it working using this one:

`U8G2_ST7567_JLX12864_F_4W_HW_SPI u8g2(U8G2_MIRROR, PIN_LCD_CS, PIN_LCD_A0, PIN_LCD_RST);`

However, the display bitmap is still flipped about both the X and Y axes unless I call 

`u8g2.setFlipMode(1);`

Is it possible to use this information to create a dedicated constructor for the SLG display? I suspect the driver is mostly the same except for the scan direction or byte order.
</div>

olikraus, 2021-11-16:
@johngineer

did you test some of the other constructors?
ST7567 ENH_DG128064
ST7567 ENH_DG128064I
will they work better?

johngineer, 2021-11-16:
`U8G2_ST7567_ENH_DG128064_F_4W_HW_SPI u8g2(U8G2_R0, PIN_LCD_CS, PIN_LCD_A0, PIN_LCD_RST);`
and calling 
`setFlipMode(1);`
works.

Using
`U8G2_ST7567_ENH_DG128064I_F_4W_HW_SPI u8g2(U8G2_MIRROR, PIN_LCD_CS, PIN_LCD_A0, PIN_LCD_RST);`
and calling 
`setFlipMode(1);`
also works.

olikraus, 2021-11-17:
But then: Wouldn't it be good enough to use U8G2_ST7567_ENH_DG128064_F_4W_HW_SPI  with setFlipMode(1) ?

johngineer, 2021-11-17:
Yes, it's not an issue for me to do just that.

I figured it would be useful to share how I got this particular display working so that others don't need to spend the time figuring it out.

johngineer, 2021-11-17:
Out of curiosity, is there a way to create a constructor that will automatically set FlipMode to 1?

olikraus, 2021-11-17:
Yes, but it requires some effort

johngineer, 2021-11-17:
I understand. Thank you for this wonderful library!

closed 2022-01-07 by olikraus