olikraus/u8g2 #1698: Internal buffer and threading

sgjava, 2021-11-18:
@olikraus I've built a simple program based on SW I2C using 3 displays and 3 threads (using arm-linux port). The good news is they all 3 fire up in a thread, but the bad news is they all seem to share the same internal buffer (i.e. I see the same sting in all 3, but they should be different). I'll look around at the internals, but do you have an idea where to start?

Using u8g2_Setup_ssd1306_i2c_128x32_univision_f

olikraus, 2021-11-18:
Correct, the displays try to share the same buffer. I think this should be ok, because threads are usually not used and also in most cases there is only one display:

https://github.com/olikraus/u8g2/blob/42d2918c0f4d7e9c1457abfe1954999a384df0ac/csrc/u8g2_d_setup.c#L1239

The buffer allocation procedure itself is here: https://github.com/olikraus/u8g2/blob/42d2918c0f4d7e9c1457abfe1954999a384df0ac/csrc/u8g2_d_memory.c#L28-L37

As you can see there is already a #ifdef . The problem you mentioned should be solved already. Just define U8G2_USE_DYNAMIC_ALLOC during compilation. This should create individual buffers.

I am not 100% whether this U8G2_USE_DYNAMIC_ALLOC will work, because I didn't wrote the code for this.

sgjava, 2021-11-18:
OK, let me test it out.

sgjava, 2021-11-18:
@olikraus could I handle this through setBufferPtr?

olikraus, 2021-11-18:
Maybe. As I said, i am not the author of this feature.
But yes, from somewhere the memory area is needed.

sgjava, 2021-11-18:
Just setting def didn't work, so I'm trying creating my own buffer and setting it.

sgjava, 2021-11-18:
OK, it's a combo deal. U8G2_USE_DYNAMIC_ALLOC and
```
	buf = (uint8_t*)malloc(u8g2_GetBufferSize(&u8g2));
	u8g2_SetBufferPtr(&u8g2, buf);
```

PR with demo coming soon. Also need to put up a youtube video running three displays in different threads. Using those little 128x32 displays could have real world applications. Only like $8 US at Amazon for 3.

closed 2021-11-18 by sgjava

olikraus, 2021-11-18:
:-D 
happily looking forward to watch youtube.

sgjava, 2021-11-18:
Yes, let me get code up and I'll do video next.