olikraus/u8g2 #1702: Support for AVR® DB family

JDJelectronics, 2021-11-18:
Hi olikraus,

Can you make support the u8g2 lib on newer AVR proccesors?

We love to use you u8g2 lib on the AVR128DB32 and AVR128DB48 proccecors. 

Greats,

JDJ ELECTRONICS

olikraus, 2021-11-19:
Not sure how I can support here. U8g2 support Arduino. If the mentioned processor is supported by Arduino, then also U8g2 will support those controllers.

On any none-Arduino environment, there is a description on how to use u8g2 on any unknown or new controller: https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform

JDJelectronics, 2021-11-19:
the proccesor is support bij arduino. 

https://github.com/SpenceKonde/DxCore

olikraus, 2021-11-19:
Well, then nothing needs to be done here. U8g2 builds on top of the Arduino Language.

JDJelectronics, 2021-11-19:
> Well, then nothing needs to be done here. U8g2 builds on top of the Arduino Language.

I got it to work sorry didn't read correctly.

I do have 1 thing:

 u8g2.firstPage();
  do {

} while ( u8g2.nextPage() );

is working fine, but the refresh rate of the oled is very slow.

send buffer command does nothing.

what could be the problem?

olikraus, 2021-11-19:
> but the refresh rate of the oled is very slow.

Might be true if the software emulated interface is active.

JDJelectronics, 2021-11-21:
Oke thanx, i will try nextweek!

I also made another icon pack of the Android 12 status bar icons for the u8g2 lib.

would you be willing to accept it and put it in the library?

I don't have all the icons yet, but I'm still working on it.

here is the first versio
![PXL_20211121_161251442](https://user-images.gith
[Statusbar_icons_android_U8G2.zip](https://github.com/olikraus/u8g2/files/7576602/Statusbar_icons_android_U8G2.zip)
ubusercontent.com/16858919/142770031-2a94d34e-68b8-42a7-a2a9-bb7e72a40917.jpg)
n.

JDJelectronics, 2021-11-21:
Okay thanx!! iw will try next week!

I also made another icon pack of the Android 12 status bar icons for the u8g2 lib.

would you be willing to accept it and put it in the library?

I don't have all the icons yet, but I'm still working on it.

here is the first version.

[image]

JDJelectronics, 2021-11-21:
[Statusbar_icons_android_U8G2.zip](https://github.com/olikraus/u8g2/files/7576606/Statusbar_icons_android_U8G2.zip)

olikraus, 2021-11-21:
Nice. Thanks. I will check the icons.

JDJelectronics, 2021-11-21:
I am updating the repo here: https://github.com/JDJelectronics/u8g2_Android_icons

olikraus, 2022-03-04:
I think there is no todo from my end, closing...

closed 2022-03-04 by olikraus