olikraus/u8g2 #1710: ST7525 JLX19264 support

Yelgurk, 2021-11-25:
In the library I found support for st7525_jlx19296 display. Can I use this setup to work with the st7525_jlx19264 display?

olikraus, 2021-11-25:
Of course you can use the existing constructor. Probably the existing constructor will not fully work, but creation of a new constructor is much easier if there is already an existing partly working constructor.
Maybe you can post a picture of the output for your display with the existing constructor. 
It might be required to adjust the contrast to see anything.
Additionally: Is there any datasheet for your display available?

olikraus, 2021-11-25:
BTW: Do you mean ST75256 controller? (missing "6" at the end?)

Yelgurk, 2021-11-25:
Display: JLX19264G-925
Controller: ST7525 (without 6)
Size: 192x64
I have the datasheet, but it's in Chinese. I uploaded it to Google Drive. Link: https://drive.google.com/file/d/1ekzxsqFYnE27WAUp0LaKjgLisw3cqK-3/view?usp=sharing

olikraus, 2021-11-25:
> Controller: ST7525 (without 6)

But you said, there is support in u8g2 for this controller. Are you sure? U8g2 became large and I may not remember all supported controllers. Any link into u8g2 code for the ST7525 support?

Yelgurk, 2021-11-25:
You're right. I was looking at 7565. My bad. There is only 8282 series. Maybe there is a misprint in the datasheet and the display uses a st75256 controller. Then you corrected me correctly and the question remains the same. I found the following setup code "u8g2_Setup_st75256_jlx19296_f", but I have no idea how to convert this setup for a smaller display.

olikraus, 2021-11-25:
This page also mentions st7525: http://www.jlxlcd.cn/html/zh-list-011014.html

olikraus, 2021-11-25:
From the ST7525 datasheet it looks similar to ST7565, except it is more wider.
Maybe you can test one of the ST7565 constructor devices. If one of them shows something, we could use this as a starting point.

olikraus, 2022-03-04:
closing for now... feel free to reopen

closed 2022-03-04 by olikraus