olikraus/u8g2 #1711: I2C  not working with Waveshare 1.5” 12x128 ssd1327 OLED

mccgregory, 2021-11-26:
I have been trying to drive a Waveshare 1.5" 128x128 OLED with SSD1327, via the I2C buss.
I failed and ended up on your u8x8/u8g2 site.
I established the Display actually worked by using it in SPI mode.
Whilst being reasonably happy about this, I would prefer to use I2C (I don't need the refresh speed) so I tried again with I2C.
If I use a very simple "Hello World" program then it fails as it doesn't fully recognise the Constructor Line.

I’m using an Arduino Pro Mini to drive this, and it works fine with a smaller OLED (0.96”) with Adafruit libraries.
I then swapped to the u8x8 "Hello World" example and picked a random ssd1306 driver Constructor Line – worked 1st time.
Every SSD1327 Constructor Line fails to Compile.
I can swap back to the working SSD1306 Constructor Line and it compiles.
Incidentally, my 1.5” OLED has its DC pin connected to Ground so it should work with the default 0x3C used by your U8x8 (when it eventually, compiles and uploads).

MY CODE is (with the different Constructor Lines I have tried - and the result) :

```
#include <Arduino.h>
#include <U8x8lib.h>

//U8X8_SSD1306_128X64_NONAME_SW_I2C u8x8(/* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE);   //worked for 0.96" ssd1306 OLEDs
//U8X8_SSD1306_128X64_NONAME_SW_I2C u8x8(SCL,SDA,U8X8_PIN_NONE);   //Worked fine
//U8X8_SSD1306_128X64_NONAME_SW_I2C u8x8(5,4,U8X8_PIN_NONE);   //Compiled, uploaded but didn't drive display
//U8X8_SSD1306_128X64_NONAME_SW_I2C u8x8(A5,A4,U8X8_PIN_NONE);   //Worked fine display

//U8G2_SSD1327_WS_128X128_1_HW_I2C(U8G2_R0);
//U8X8_SSD1327_WS_128X128_SW_I2C(A4,A5);

//U8X8_SSD1327_WS_128X128_HW_I2C(U8X8_PIN_NONE);    //Needs U8X8_PIN_NONE, but doesn't compile
//U8X8_SSD1327_WS_128X128_HW_I2C(U8X8_PIN_NONE,A4,A5);    //Needs U8X8_PIN_NONE, but doesn't compile
//U8X8_SSD1327_MIDAS_128X128_HW_I2C(U8X8_PIN_NONE,A4,A5);    //Needs U8X8_PIN_NONE, but doesn't compile

//U8X8_SSD1327_EA_W128128_HW_I2C([reset [, clock, data]]);
//U8X8_SSD1327_EA_W128128_HW_I2C(U8X8_PIN_NONE,);   //Needs U8X8_PIN_NONE, but doesn't compile

//U8X8_SSD1327_EA_W128128_SW_I2C(clock, data [, reset])
//U8X8_SSD1327_EA_W128128_SW_I2C(SCL,SDA,U8X8_PIN_NONE);   //Needed U8X8_PIN_NONE + something else, or  "expected unqualified-id before numeric constant"  No Compile

//U8X8_SSD1327_MIDAS_128X128_HW_I2C([reset [, clock, data]])
//U8X8_SSD1327_MIDAS_128X128_HW_I2C(U8X8_PIN_NONE); //Needs U8X8_PIN_NONE, but doesn't compile

//U8X8_SSD1327_WS_128X128_HW_I2C([reset [, clock, data]])
//U8X8_SSD1327_WS_128X128_HW_I2C(U8X8_PIN_NONE);   //Needs U8X8_PIN_NONE, but doesn't compile

//U8X8_SSD1327_WS_128X128_SW_I2C(clock, data [, reset])
U8X8_SSD1327_WS_128X128_SW_I2C(A5,A4,U8X8_PIN_NONE);   //Needed U8X8_PIN_NONE + something else, or  "expected unqualified-id before numeric constant"  No Compile

void setup(void)
{
  u8x8.begin();
  u8x8.setPowerSave(0);
}

void loop(void)
{
  u8x8.setFont(u8x8_font_chroma48medium8_r);
  u8x8.drawString(0,1,"Hello World!");
  delay(2000);
}

```

If I use, for example, U8X8_SSD1327_WS_128X128_HW_I2C(U8X8_PIN_NONE); then it fails at the first following line:
i.e.    u8x8.begin();
with an error message: 'u8x8' was not declared in this scope

If I change this to a Constructor Line for SPI, then the CODE compiles.
I have tried several different Constructor Lines with the same result.
One interesting thing is it initially failed with:
                "expected unqualified-id before numeric constant".
                I fixed this by changing the line:
                from      U8X8_SSD1327_WS_128X128_HW_I2C();
                to:                          U8X8_SSD1327_WS_128X128_HW_I2C(U8X8_PIN_NONE);
                Which, surely, tells me it can find the Libraries and U8x8lib.h.
                
Another point of interest- when I compile with verbose it shows:
                "Using library SPI at version 1.0 in folder: C:\Program Files"
Why would it call SPI.h when I don't, and logically it doesn't require it?

I have spent several hours struggling with this, reading the wiki and the FAQ etc. and would appreciate some guidance.

Having said all that, I’ve just seen “support for SSD1327 128x128 1.5" I2C #684”
in there you say:
“Which uC do you use? 128x128 will require 2KBytes: This will be too much for any ATMega3xx based system like Uno, Mini, Nano or similar”.

Although I haven’t seen any memory warnings, is this my problem?

Would swapping to the u8g2 Constructor Lines with matching code fix it using memory pages?

Thanks in advance.
Greg

olikraus, 2021-11-26:
The error

> i.e. u8x8.begin();
> with an error message: 'u8x8' was not declared in this scope

indicates that the object "u8x8" is unknown and needs to be declared. 
In fact a constructor itself is just a blueprint for an object, that needs to be created by using a specific constructor: If a constructor is used as a variable type (which is true for most of the Arduino libs) then the object itself (the thing which needs to be constructed) needs to be prefixed by the constructor name. In U8g2/U8x8 lib the objects are either called u8g2 or u8x8 (actually this name doesn't matter, but i like those two object names). So... it should look like this:

<constructor> u8x8

Unfortunately the constructor needs additional arguments and even more wired, these arguments for the constructor have to be passed as arguments to the object. I have not invented this, please complain @ C++ people. So it will look like this:

<constructor> u8x8(arg 1, arg 2, ...)

So you can say, that <constructor>  and (arg 1, arg 2, ...) belong together, but the object name (u8x8) sits in the middle.

Lets have a look at you constructor lines:
```
//U8X8_SSD1306_128X64_NONAME_SW_I2C u8x8(SCL,SDA,U8X8_PIN_NONE);   //Worked fine
```

Looks good and as you stated, works fine: `U8X8_SSD1306_128X64_NONAME_SW_I2C` is the constructor name, `u8x8` the required object name (which will be declared here) and `(SCL,SDA,U8X8_PIN_NONE)` are the extra args for the constructor.

Now lets have a look at your SSD1327 lines:
```
//U8X8_SSD1327_WS_128X128_SW_I2C(A4,A5);
```
You see the problem?
It must be:
```
//U8X8_SSD1327_WS_128X128_SW_I2C u8x8(A4,A5);
```

mccgregory, 2021-11-27:
Thank you for your rapid and patient response.

I see what you mean.

It took me several minutes of staring but I got it in the end.

 

My problem is basically I’m a hobbyist riding on the back of coding giants like yourself – I don’t understand what I’m doing but I bumble along with the project.

Apologies for my foolishness.

But genuinely thanks for your patience.

 

I would humbly say though that when I copied, or used a Constructor Line e.g. 

U8X8_SSD1327_EA_W128128_HW_I2C([reset [, clock, data]]);

I didn’t realise I would also need to add u8x8 as below:

U8X8_SSD1327_EA_W128128_HW_I2C u8x8([reset [, clock, data]]);

As well as fill in the arguments! 😊

My old, foolish eyes were concentrated on the interior of the parenthesis trying to get the correct details in the right order.

 

Right Onwards and Upwards!

The lines now compile, the CODE uploads but, as yet, the OLED doesn’t respond.

You can close that call, but I may well be back.

 

Again, a warm Thank YOU to you Olikraus?  

Greg

 

u8x8(arg 1, arg 2, ...)

So you can say, that and (arg 1, arg 2, ...) belong together, but the object name (u8x8) sits in the middle.

olikraus, 2021-11-27:
I fully understand your confusion. It is indeed confusing. Both version (with and without object name in the middle) are valid but depend on the context: https://www.w3schools.com/cpp/cpp_constructors.asp

closed 2022-03-04 by olikraus