olikraus/u8g2 #1712: Support for AVRs with two hardware i2c ports

MCUdude, 2021-11-28:
Hi!
There has been reported over at the [MiniCore repo](https://github.com/MCUdude/MiniCore/issues/204) that u8g2 doesn't compile when the ATmega328PB is selected. The same is true for MightyCore + ATmega324PB too.

I can easily recreate the issue by compiling the Arduino FullBuffer/HelloWorld example, and using the `U8G2_SSD1306_128X64_ALT0_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);` constructor.

Here is the error:

```
/Users/hans/Documents/Arduino/libraries/U8g2/src/U8x8lib.cpp: In function 'uint8_t u8x8_byte_arduino_2nd_hw_i2c(u8x8_t*, uint8_t, uint8_t, void*)':
/Users/hans/Documents/Arduino/libraries/U8g2/src/U8x8lib.cpp:1355:7: error: 'Wire1' was not declared in this scope
       Wire1.write((uint8_t *)arg_ptr, (int)arg_int);
       ^~~~~
/Users/hans/Documents/Arduino/libraries/U8g2/src/U8x8lib.cpp:1355:7: note: suggested alternative: 'Wire'
       Wire1.write((uint8_t *)arg_ptr, (int)arg_int);
       ^~~~~
       Wire
```

The reason for this is because Wire1.h has to be included in order for the second hardware i2c port to work.
However, when Wire1.h is included, the memory usage is increased. The sketch compiles down to 9566 bytes on an ATmega328P, but ... on the ATmega328PB. Since the second hardware serial port isn't much used anyways on these AVRs, is there a way to not include Wire1 and still make the code compile on ATmega324PB and ATmega328PB?

olikraus, 2021-11-28:
You could comment this line:
https://github.com/olikraus/u8g2/blob/956507bc912b58cb66ea851c94b787a28de08db8/cppsrc/U8x8lib.h#L117

MCUdude, 2021-11-28:
Thanks, this works.

> By the way: There should be no Wire1.h, instead Wire1 should be part of Wire.h according to my understanding.

I've also tried to inculde Wire1.h in Wire.h, but this also makes the compiled code much larger. For some reason the compiler aren't able to optimize out Wire1 even though it's not in use. The same goes for Wire. Do you know of any tricks that makes the compiler optimize away libraryes (like Wire) that has been included but are not in use?

olikraus, 2021-11-28:
> Serial/Serial1/Serial2/Serial3 works this way, but I have not figured out how it's done.

Yes, I also noticed this. I think the problem is, that Wire is defined in the C++ file of the library:
```
TwoWire Wire = TwoWire();
```
The Wire constructor is called as part of the global init and can not be removed by the linker garbage collector.
SPI is different: It does not have a constructor function (as far as I know). This allows the linker garbage collector to remove the SPI object.

I think Wire is a bad example where the Arduino concept oversimplified things: Arduino project removed the need for declaration. Instead they just did the declaration for the users with the effect that too much of flash and RAM is consumed in cases where the object is not required.

```
SPIClass SPI;
```

closed 2022-03-04 by olikraus