olikraus/u8g2 #1713: any way to load this fonts in U8x8?

fdallera, 2021-11-28:
when i print it the char is half cut

const uint8_t DEDFont[1796] U8X8_FONT_SECTION("DEDFont") = 
  " \377\1\1\0\0\0\0\0\0\0\0\0\0\0\15\15\0\0\0\0\0\0\0\0\0\0\0\0\2\17\17"
  "\2\17\17\2\0\4\14\70\71\17\6\0\0\6\3\15\36\22\36\14\0\7\17\10\11\7\17\11\0\0\0\0"
  "\0\0\0\0\0\0\7\37\70 \0\0\0\0 \70\37\7\0\0\37\37\35\34\36\34\35\37\0\0\0\3"
  "\3\0\0\0\0\0\0,<\34\0\0\0\0\0\0\0\0\0\0\0\0\0\14\14\14\0\0\0\30\36\7"
  "\1\0\0\0\0\0\7\17\13\10\17\7\0\0\0\0\17\17\0\0\0\14\16\13\11\10\10\0\0\6\16\10"
  "\10\17\7\0\0\3\3\2\2\17\17\2\0\10\10\10\14\7\3\0\0\7\17\10\10\17\7\0\0\0\16\17"
  "\1\0\0\0\0\7\17\10\11\17\7\0\0\0\11\15\17\3\1\0\0\0\0\14\14\14\0\0\0\0\0,"
  "<\34\0\0\0\0\1\3\6\14\10\0\0\1\1\1\1\1\1\0\0\10\14\6\3\1\0\0\0\0\0\15"
  "\15\0\0\0\7\17\10\11\13\12\13\13\0\17\17\1\1\17\17\0\0\17\17\10\10\17\7\0\0\7\17\10"
  "\10\16\6\0\0\17\17\10\14\7\3\0\0\17\17\10\10\10\10\0\0\17\17\0\0\0\0\0\0\7\17\10"
  "\11\17\17\0\0\17\17\0\0\17\17\0\0\0\10\17\17\10\0\0\0\6\16\10\10\17\7\0\0\17\17\0"
  "\3\17\14\0\0\17\17\10\10\10\10\0\0\17\17\0\1\0\17\17\0\17\17\0\0\1\17\17\0\7\17\10"
  "\10\17\7\0\0\17\17\0\0\0\0\0\0\7\17\10\30\77'\0\0\17\17\0\1\17\16\0\0\4\14\10"
  "\11\17\6\0\0\0\0\17\17\0\0\0\0\7\17\10\10\17\7\0\0\3\7\14\14\7\3\0\0\1\17\16"
  "\1\16\17\1\0\16\17\0\1\17\16\0\0\0\0\17\17\0\0\0\0\16\17\11\10\10\10\0\0\0\177\177"
  "@@\0\0\1\17\16\1\16\17\1\0\0\0@@\177\177\0\0\0\0\0\0\0\0\0\0@@@@"
  "@@@@\0\0\0\0\0\0\0\0\0\1\3\7\17\7\3\1\0\17\17\10\10\17\7\0\0\7\17\10"
  "\10\14\4\0\0\7\17\10\10\17\17\0\0\7\17\11\11\11\1\0\0\0\17\17\0\0\0\0\0GOH"
  "H\177\77\0\0\17\17\0\0\17\17\0\0\10\10\17\17\10\10\0\0@@@\177\77\0\0\0\17\17\1"
  "\3\16\14\0\0\10\10\17\17\10\10\0\0\17\17\0\7\0\17\17\0\17\17\0\0\17\17\0\0\0\0\0"
  "\0\0\0\0\0\177\177\10\10\17\7\0\0\7\17\10\10\177\177\0\0\17\17\0\0\0\0\0\0\10\11\11"
  "\11\17\6\0\0\0\7\17\10\10\10\0\0\7\17\10\10\17\17\0\0\3\7\14\14\7\3\0\0\3\17\14"
  "\3\14\17\3\0\14\16\3\3\16\14\0@GOh\70\37\7\0\0\14\16\13\11\10\10\0\0\1\3\36"
  "< \0\0\0\0\0\177\177\0\0\0\0\0 <\36\3\1\0\0\0\0\0\0\0\0\0\0\17\17\17"
  "\17\17\17\0\0\17\17\17\17\17\17\0\0\17\17\17\17\17\17\0\0\17\17\17\17\17\17\0\0\17\17\17"
  "\17\17\17\0\0\17\17\17\17\17\17\0\0\17\17\17\17\17\17\0\0\17\17\17\17\17\17\0\0\17\17\17"
  "\17\17\17\0\0\17\17\17\17\17\17\0\0\17\17\17\17\17\17\0\0\17\17\17\17\17\17\0\0\17\17\17"
  "\17\17\17\0\0\17\17\17\17\17\17\0\0\17\17\17\17\17\17\0\0\17\17\17\17\17\17\0\0\17\17\17"
  "\17\17\17\0\0\17\17\17\17\17\17\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\17\17\17"
  "\17\17\17\0\0\17\17\17\17\17\17\0\0\17\17\17\17\17\17\0\0\17\17\17\17\17\17\0\0\17\17\17"
  "\17\17\17\0\0\17\17\17\17\17\17\0\0\17\17\17\17\17\17\0\0\17\17\17\17\17\17\0\0\17\17\17"
  "\17\17\17\0\0\17\17\17\17\17\17\0\0\17\17\17\17\17\17\0\0\17\17\17\17\17\17\0\0\17\17\17"
  "\17\17\17\0\0\0\0\0\0\0\0\0\0\0\70\376\376\70\0\0\0\3\7\34\34\6\2\0\14\17\13\10"
  "\10\10\10\0\0\2\3\1\1\3\2\0\0\2\2\17\17\2\2\0\0\0\0\374\374\0\0\0\0#\347\314"
  "\314\377\63\0\0\0\0\0\0\0\0\0\7\14\11\12\12\11\14\7\0\2\2\2\2\2\2\0\2\7\15\10"
  "\2\7\15\10\0\0\0\0\0\3\3\0\0\0\0\0\0\0\0\0\7\14\13\11\11\12\14\7\0\0\0\0"
  "\0\0\0\0\0\0\0\0\0\0\0\0\0\10\10\13\13\10\10\0\0\0\0\0\0\0\0\0\0\0\0\0"
  "\0\0\0\0\0\0\0\0\0\0\0\0\300\377\77\10\10\7\17\10\0\0\1\3\3\377\377\0\0\0\0\0"
  "\0\0\0\0\0\0\300\320\360 \0\0\0\0\0\0\0\0\0\0\0\2\2\2\2\2\2\0\10\15\7\2"
  "\10\15\7\2\4\6\3\15\16\13\37\37\4\6\3\1\31\35\27\22\4\6\3\15\16\13\37\37\0\60\370\316"
  "\306\360\60\0\0\17\17\1\1\17\17\0\0\17\17\1\1\17\17\0\0\17\17\1\1\17\17\0\0\17\17\1"
  "\1\17\17\0\0\17\17\1\1\17\17\0\0\17\17\1\1\17\17\0\0\17\17\1\17\17\10\10\0\7\317\330"
  "\370.\6\0\0\17\17\10\10\10\10\0\0\17\17\10\10\10\10\0\0\17\17\10\10\10\10\0\0\17\17\10"
  "\10\10\10\0\0\0\10\17\17\10\0\0\0\0\10\17\17\10\0\0\0\0\10\17\17\10\0\0\0\0\10\17"
  "\17\10\0\0\0\17\17\10\14\7\3\0\0\17\17\0\0\1\17\17\0\7\17\10\10\17\7\0\0\7\17\10"
  "\10\17\7\0\0\7\17\10\10\17\7\0\0\7\17\10\10\17\7\0\0\7\17\10\10\17\7\0\0\6\3\1"
  "\1\3\6\0\0\17\17\13\10\17\7\0\0\7\17\10\10\17\7\0\0\7\17\10\10\17\7\0\0\7\17\10"
  "\10\17\7\0\0\7\17\10\10\17\7\0\0\0\0\17\17\0\0\0\0\17\17\2\2\3\1\0\0\17\17\0"
  "\10\17\7\0\0\6\17\11\11\17\17\0\0\6\17\11\11\17\17\0\0\6\17\11\11\17\17\0\0\6\17\11"
  "\11\17\17\0\0\6\17\11\11\17\17\0\0\6\17\11\11\17\17\0\6\17\11\17\7\11\11\11\0\7\317\330"
  "\370,\4\0\0\7\17\11\11\11\1\0\0\7\17\11\11\11\1\0\0\7\17\11\11\11\1\0\0\7\17\11"
  "\11\11\1\0\0\10\10\17\17\10\10\0\0\10\10\17\17\10\10\0\0\10\10\17\17\10\10\0\0\10\10\17"
  "\17\10\10\0\0\7\17\10\10\17\7\0\0\17\17\0\0\17\17\0\0\7\17\10\10\17\7\0\0\7\17\10"
  "\10\17\7\0\0\7\17\10\10\17\7\0\0\7\17\10\10\17\7\0\0\7\17\10\10\17\7\0\0\0\0\6"
  "\6\0\0\0\0\17\17\13\11\17\7\0\0\7\17\10\10\17\17\0\0\7\17\10\10\17\17\0\0\7\17\10"
  "\10\17\17\0\0\7\17\10\10\17\17\0\300\307\317\350\70\37\7\0\0\377\377\10\10\17\7\0\300\307\317\350"
  "\70\37\7";

olikraus, 2021-11-29:
Any pictures? More code?

fdallera, 2021-12-01:
this is a 16 bit font i would like to load it in 8x8 
in 8x8 there is an example of 16bit font loaded i didn't find any difference in header definition but it doesn't work

fdallera, 2021-12-01:
[image]

olikraus, 2021-12-02:
Somehow it is not clear to me what you mean. Please provide an code example. Describe or show what you get and describe what you would expect instead.

closed 2022-03-04 by olikraus