olikraus/u8g2 #1715: Platformio version is very old

positron96, 2021-12-03:
Hi. 
Could you please bump the version of the library in platformio registry? Current is 2.28.0 which is from 2020. 
https://platformio.org/lib/show/942/U8g2

olikraus, 2021-12-03:
I know my name is mentioned on platformio, but I have never added u8g2 to platformio. I do not even have a login for platformio.org. 

It had been the platformio project which had linked u8g2 to their project, so please ask the maintainer of u8g2 on platformio.org to update to the latest library. There is nothing what I can do from my end.

positron96, 2021-12-03:
Thanks for the response! I'll try to...

positron96, 2021-12-04:
So, here are the results. 
As I understand it, u8g2 library is in pio due to automatic processing of official Arduino repository. 
I've [reported](https://community.platformio.org/t/library-update-not-picked-up-by-platformio-org/1294/19?u=positron96) the problem with the library not updating, pio guys promptly [updated](https://github.com/platformio/platformio-core/commit/aab42c3cffeabc542d6821bdb55c7a4743a803f7) their parser and now the library is up-to-date. Apparently it had something to do with long fields in `library.properties`.

closed 2021-12-04 by positron96

olikraus, 2021-12-04:
Thanks for the feedback. Is there something I can improve about the U8g2 library.properties file?

https://github.com/olikraus/U8g2_Arduino/blob/f014fc898afb13460323243588a17a73043aecc2/library.properties#L1-L2

positron96, 2021-12-07:
Sorry, I have no idea. 
Judging from [spec](https://arduino.github.io/arduino-cli/0.19/library-specification/), there is nothing wrong with syntax of the `library.properties`, no limit is put on the length of the description.

Though it states that `sentence` should be one sentence and `paragraph` should continue, not repeat the sentence (which u8g2 
 library violates). But I'm sure it does not prevent correct parsing of the file.

olikraus, 2021-12-07:
Hmmm