olikraus/u8g2 #1717: I try to write SSD1681 driver (e-paper) to u8d2 and I need some help

pikot, 2021-12-03:
Hi, I'm adapting now u8g2 library to work with e-paper display with SSD1681 driver, and it shows signs of life =) code for driver is [there](https://github.com/pikot/U8g2_Arduino/blob/master/src/clib/u8x8_d_ssd1681_200x200.c) 
Sometimes all work fine as on photo
[image]

sometimes it works with black pixels on screen
[image]

I can't understand why it's happened. I tried to change some delays, but all doesn't have effect.  It's my first experience with displays drivers and e-ink especially. 
Can somebody give any hint where I should look info to fix this problem (It's definitely driver problem, some pixels don't reseted its color, and I can't understand why i have broken picture) 

btw. delays that used instead reading busy signal status, it's creepy, why library avoid busy line? ))

olikraus, 2021-12-04:
> I can't understand why it's happened. I tried to change some delays, but all doesn't have effect. It's my first experience with displays drivers and e-ink especially.

Probably you need the correct LUT for your display. The LUT is the secret for proper e-paper usage: For example the LUT for the SSD1607 starts here: 
https://github.com/olikraus/u8g2/blob/5c7d5da050624c2e1fb9edbd34966cc62f149ae4/csrc/u8x8_d_ssd1607_200x200.c#L249

Explaining the LUT is difficult, but basically this is a sequence of commands to change a pixel value. If i remember correctly, there will be a command sequence for all possible transitions: white pixel to white, white to black, black to black and black to white. The commands are usually something  like this: "apply certain voltage for a specific time" to the pixel.

LUT differ between e-paper controllers, so careful study of the e-paper controller is required.

> btw. delays that used instead reading busy signal status, it's creepy, why library avoid busy line? ))

This is one of the reasons, why I have up supporting e-paper with u8g2. Probably the famos epaper lib from @ZinggJM is much better here. In fact @ZinggJM  might support the SSD1681 already.
On the other side: The programming time (indicated by the busy line) depends on the LUT and is fixed if the LUT doesn't change, so a busy delay seems to be ok from my end.

olikraus, 2021-12-04:
I really like your code (looks like you studied u8g2 a lot), but indeed the LUT setup is missing.
Loading the LUT from internal memory might work, but you never know whether the predefined LUT is correct: 
https://github.com/pikot/U8g2_Arduino/blob/6ba90554449cb74b19967a88d4c6e154ca3a752d/src/clib/u8x8_d_ssd1681_200x200.c#L140

I think all your problems will be solved if you download your own optimized LUT to the controller.

ZinggJM, 2021-12-04:
In https://github.com/ZinggJM/GxEPD2/blob/master/examples/GxEPD2_Example/GxEPD2_display_selection_new_style.h
```
#define GxEPD2_DRIVER_CLASS GxEPD2_154_D67 // GDEH0154D67 200x200, SSD1681
``` 
Note that GDEH0154D67 from Good Display has waveform table in OTP for full and for differential refresh.

pikot, 2021-12-04:
> This is one of the reasons, why I have stopped supporting e-paper with u8g2. Adding the busy line to u8g2 would have been very difficult. 

I have idea about busy line. As I understand, cad FSM in delay state calls u8x8_gpio_and_delay (u8x8_gpio_and_delay_arduino) callback, and  sounds reasonable to insert busy line  monitoring there.  
As I see 
1. in U8G2_SSD1681_200X200_*_*W_*W_SPI constructor add new argument busy 
2. in u8x8_SetPin_*Wire_*W_SPI function call  u8x8_SetPin(u8x8, U8X8_PIN_BUSY, busy);  (yep, here busy line in most cases may be unused, and for displays which use busy line I should write separate functions u8x8_SetPin_4WireB_SW_SPI (for example)
3.  in u8x8_gpio_and_delay_arduino I have access to (u8x8)->pins[pin]   and in case  U8X8_MSG_DELAY_MILLI  if  (u8x8)->pins[U8X8_PIN_BUSY] not some default value, I can do busy line monitoring function and arg_int use as timeout value.
May be I can't see some difficulties due to I'm not so proficient in u8g2 project, but busy line logic monitoring looks clear and transparent here

 > Probably the famos epaper lib from @ZinggJM is much better here. In fact @ZinggJM might support the SSD1681 already.
I have project which works with oled via u8g2 and switching to @ZinggJM 's library problematic ) It's easier to me write new driver for u8g2, and it is fun)

olikraus, 2021-12-04:
> It's easier to me write new driver for u8g2

Well, then maybe you should at least takeover the setup sequence from GDEH0154D67

pikot, 2021-12-04:
> I really like your code (looks like you studied u8g2 a lot), but indeed the LUT setup is missing. Loading the LUT from internal memory might work, but you never know whether the predefined LUT is [...]

According to datasheet, section "8.1 General operation flow to drive display panel" I can load  waveform LUT from OTP by Command 0x22, 0x20 (@ZinggJM  mention it) and I thought command 0x22, 0xb1 (Load LUT with DISPLAY Mode 1) thats what I need, may be it's not correct, I will look in datasheet again  )   

thanks for hint!

pikot, 2021-12-28:
hi! afrer some tests and implementing and soldering in real live pcb for D57 display I understund that problem was in LUT which I load from OPT memory based on constant temperature =) after that I set reading internal sensor and all looks fine)

btw! I implemented scanning BUSY pin logic in library ) now I do extra tests and debugging I'm going to push request (with busy logic and ssd1681 driver, at first  with full update feature)

closed 2022-03-04 by olikraus

jacky4566, 2022-05-12:
@pikot Were you going to make a merge request? I would love to help with the ssd1681 driver.

pikot, 2022-05-12:
My code is here https://github.com/pikot/U8g2_Arduino , I need some time to copy that into U8g2 repo, I will try to do it this weekend