olikraus/u8g2 #1720: SSD1327 and ESP C3

elgerg, 2021-12-10:
Hi,

I have a SSD1327 (http://www.lcdwiki.com/zh/1.5inch_OLED_Module_SKU:MSP1503) and a ESp32 C3 S2 (https://www.waveshare.com/wiki/ESP-C3-32S-Kit).

I have tested that the I2C is working using the Adafruit I2C tester and it confirms that there is a device on 0x3C:
Adafruit_I2CDevice i2c_dev = Adafruit_I2CDevice(0x3C);

However, I cant get the screen to show anything. Im probably using the wrong constructor. EG:
U8G2_SH1107_SEEED_128X128_1_SW_I2C u8g2(U8G2_R0, /* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE);
or
U8GLIB_SSD1327_96X96_2X_GR u8g(U8G_I2C_OPT_NONE);  //IIC
or
U8X8_SSD1327_EA_W128128_2ND_HW_I2C u8g2(U8X8_PIN_NONE);

If anyone has any ideas on how to get this working I would really appreciate it.

SCL is connected to IO9 and SDA is connected to IO8.

Thanks in advance
Alex

olikraus, 2021-12-10:
Did you also test  U8X8_SSD1327_EA_W128128_SW_I2C?

The U8G2_SH1107_SEEED_128X128_1_SW_I2C will not work, SH1107 is a different controller.

elgerg, 2021-12-10:
Wow, thanks for the prompt response.

I tried;
U8X8_SSD1327_EA_W128128_SW_I2C u8x8(/* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE);

But no luck, Any other suggestions?

Thanks!

olikraus, 2021-12-10:
Did you connect the display to the right pins?
I mean you have used SCL and SDA in the constructor, but you wrote that you have connected the display to IO9 and IO8.
U8g2 expects a pin number, so if SCL is defined as 11, then it probably will not work. Also IO9 is not a pin number.
U8g2 needs to know the the "Arduino" pin number...

elgerg, 2021-12-10:
I believe that for the ESP32 C3 the pin numbers for IOx and Arduino X are the same (I'd love to be proven wrong but I havent managed to find documentation on it yet). EG 3,4,5,19 and 18 all map to the same in terms of IO and Arduino.

Ive tried:
U8X8_SSD1327_EA_W128128_SW_I2C u8x8(/* clock=*/ 8, /* data=*/ 9, /* reset=*/ U8X8_PIN_NONE);
and
U8X8_SSD1327_EA_W128128_SW_I2C u8x8(/* clock=*/ 9, /* data=*/ 8, /* reset=*/ U8X8_PIN_NONE);

But no luck.

olikraus, 2021-12-10:
> EG 3,4,5,19 and 18 all map to the same in terms of IO and Arduino.

Then maybe connect your display to known pins and change the numbers in the constructor.

elgerg, 2021-12-10:
Ok, Ive tried that with pin 3 and 4. No luck.
I can seem something is happening as there is an onboard LED on PIN 3 and 4 and one if them is intermittently flashing.
I'm starting to wonder if there is something wrong with the display? 

What is the default I2C address thats used? The one I have is 0x78 (AKA 0x3C) and if thats not the default how do I override it?

Thanks again

olikraus, 2021-12-10:
https://github.com/olikraus/u8g2/wiki/u8g2reference#seti2caddress
https://github.com/olikraus/u8g2/wiki/u8x8reference#seti2caddress

elgerg, 2021-12-10:
Ok, so i added:

  u8x8.setI2CAddress(0x78);
before
  u8x8.begin();

but no luck still.
I'll have to try it with one of my old 8266 or 32's and see if the screen is broken or not..

If it does work then I guess something is not right with the C3?

Thanks for the help
Alex

olikraus, 2021-12-10:
I also assume that there is a kind of a hardware / wiring issue. SSD1327 based displays work reliably in my lab.

closed 2022-03-04 by olikraus