olikraus/u8g2 #1723: support for Raspberry Pi Pico & 2.23" waveshare OLED HAT

yo2ldk, 2021-12-11:
Hello,
First, Thank you for all your hard work !
 I have a Pi Pico module and Waveshare 2.23" 128x32 SPI/I2C OLED display,
 [https://www.waveshare.com/pico-oled-2.23.htm](url)
  changing the line (from example) for SSD1305 with :
  `U8G2_SSD1305_128X32_NONAME_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 7, /* data=*/ 6, /* cs=*/ 9, /* dc=*/ 8, /* reset=*/ 12);
  but screen is black`
  Can you help me please?
  TNX in advance !

olikraus, 2021-12-11:
Which programming environment do you use?

Maybe the pin numbers are wrong.

yo2ldk, 2021-12-11:
I programmed Raspberry Pico under Arduino, and modify noname line for ssd1305 128x32 with pin numbers for waveshare like in previous example,instead your original line.

olikraus, 2021-12-11:
In most similar cases pin numbers are wrong. How else can I help? I don't know pi pico.

yo2ldk, 2021-12-11:
.ok, thank you for your time
all the best !

closed 2021-12-11 by yo2ldk

michaelosthege, 2023-06-24:
Information for the next person struggling with the Waveshare Pico 2.23" → go to https://github.com/olikraus/u8g2/issues/2002#issuecomment-1605356307 for the solution.