olikraus/u8g2 #1728: The start line command of st7565_jlx12864 is not correct

ueJone, 2021-12-18:
```
u8g2_Setup_st7565_jlx12864_f()->
  u8x8_d_st7565_jlx12864()->u8x8_d_st7565_nhd_c12864()->
    u8x8_cad_SendSequence(u8x8, u8x8_d_st7565_nhd_c12864_init_seq)

u8x8_d_st7565_nhd_c12864_init_seq [] = {
  ...
  U8X8_C(0x40),		                /* set display start line to 0 */
  ...
}

```

st7565_jlx12864 should uses U8X8_C(0x60) to set display start line to 0

olikraus, 2021-12-18:
The start line was correct in history. Maybe you own a different version of the jlx Display. Will a different constructor for the st7565 work better?

ueJone, 2021-12-19:
> The start line was correct in history. Maybe you own a different version of the jlx Display. Will a different constructor for the st7565 work better?

Everything is fine except the position of the first line.
The model of this display device is JLX12864G-542-PN and it uses ST7565R chip.

olikraus, 2021-12-19:
We have created this constructor for a JLX12864 display, but maybe for a different model version.
I could create an alternative constructor, with a different start line. Would this be ok?

olikraus, 2022-03-04:
no further feedback... closing

closed 2022-03-04 by olikraus