olikraus/u8g2 #1729: Invert screen capture.

jakplugg, 2021-12-19:
Thank you for your great library! 

With OLED screens the screen capture results in black on white XBM file rather than white on black as it is on the screen. Is it possible to add an option to invert the bits of the output?

Thanks, JM

olikraus, 2021-12-19:
Just before writing the image, you could set the draw color to 2 (https://github.com/olikraus/u8g2/wiki/u8g2reference#setdrawcolor) and draw a box over the complete display. This would invert the image.

jakplugg, 2021-12-19:
Thats actually quite nice, the screen "flashes" to indicate a picture was sent.
thanks!

olikraus, 2021-12-19:
😃, well yes, but you probably could avoid the flashing by not sending the buffer to the display or by drawing the box again to revert the inversion.

closed 2022-03-04 by olikraus