olikraus/u8g2 #1732: Wrong paramter used in u8g2_DrawUTF8Line(..., y, ...)?

spachner, 2021-12-22:
```
static u8g2_uint_t u8g2_draw_selection_list_line(u8g2_t *u8g2, u8sl_t *u8sl, u8g2_uint_t y, uint8_t idx, const char *s)
{
  u8g2_uint_t yy;
  ...
  u8g2_DrawUTF8Line(u8g2, MY_BORDER_SIZE, y, u8g2_GetDisplayWidth(u8g2)-2*MY_BORDER_SIZE, s, border_size, is_invert);
}
```

Probably the parameter y should read yy, otherwise the local variable yy is unused?

regards

spachner

olikraus, 2021-12-23:
https://github.com/olikraus/u8g2/blob/61c71b28b116129e01d8750fd90d4d94fbe1ca0d/csrc/u8g2_selection_list.c#L151-L164

yy is not unused, it is assigned in line 160.

spachner, 2021-12-23:
Hi,
thanks for you quick reply.

I meant the variable yy is not read at all. Either yy's value is not needed and can be removed. Or yy is not read/used by accident. 

spachner

olikraus, 2021-12-23:
ah, I see your point. Hmm I assume yy and the corresponding calculation is useless and not required. It can be removed.
Nice finding, thanks!

closed 2022-01-07 by olikraus