olikraus/u8g2 #1733: Corruption on other I2C Devices after display initialise

arjus64, 2021-12-23:
I have a device on 0x78 & 0x79 and after initialising the display using the u8g2 library (display at 0x3D) all values from 0x78 & 0x79 are corrupt.

Any ideas

olikraus, 2021-12-23:
Maybe this: https://github.com/olikraus/u8g2/blob/61c71b28b116129e01d8750fd90d4d94fbe1ca0d/doc/faq.txt#L9

arjus64, 2021-12-29:
Hi,

 

Thanks for that, unfortunately I am using HW_I2C

 

The display is set at 0x3D (0x7A)

 

My pressure sensor is at 0x78 & 0x79

 

Stopping display addressing give correct pressure values but addressing and updating the display corrupts the pressure values.

 

Any ideas where I could look

 

Many Thanks

Andrew

olikraus, 2021-12-29:
Maybe the sensor is not able to work as fast as the display and the sensor lib does not restore the sensor I2C speed.
In this case you need to reduce the speed for the display: https://github.com/olikraus/u8g2/wiki/u8g2reference#setbusclock
Or assign a lower speed for the sensor as soon as you need the data.

closed 2022-03-04 by olikraus