olikraus/u8g2 #1738: u8g2_Userinterfaceselectionlist() does not support Chinese fonts well

ueJone, 2021-12-26:
[image]

```
u8g2_SetFont(&u8g2, u8g2_font_wqy12_t_gb2312);
u8g2_UserInterfaceSelectionList(&u8g2, "Title:设置", 1, "1.确认\n2.取消\n3.返回");
```
There are two questions:
1. The upper part of the title is not displayed. It use `8g2_GetAscent()` to get the height of title, but it only applies to ASCII.
2. The text of the options exceeds the background.

olikraus, 2021-12-26:
You probably need to set a different ascent/descent height: https://github.com/olikraus/u8g2/wiki/u8g2reference#setfontrefheightall
Calling
`u8g2_SetFontRefHeightExtendedText` or `setFontRefHeightAll`
after `u8g2_SetFont` may work.

ueJone, 2021-12-27:
> You probably need to set a different ascent/descent height: https://github.com/olikraus/u8g2/wiki/u8g2reference#setfontrefheightall Calling `u8g2_SetFontRefHeightExtendedText` or `setFontRefHeightAll` after `u8g2_SetFont` may work.

Perfect, thank you!

closed 2022-03-04 by olikraus