olikraus/u8g2 #1741: Compile Error for Mega 2560 says #include "mud.h" No such file

Makethings371, 2022-01-03:
Currently its "mui.h" should be it be <mui.h>? tried it several different ways still get the same same issue. Added the library and shows up in contributed library section. Any help is appreciated.

Makethings371, 2022-01-03:
Didn't mention the root file, its MUIU8g2.h 43:10 which mentions the two lines below that give a compile error.

#include "mui.h"
#include "mui_u8g2.h"

Makethings371, 2022-01-03:
I just changed it to 

#include "clib/mui.h"
#include "clib/mui_u8g2.h"

that seemed to move it to the next issue now it says root file u8g_com_HAL_AVR_sw_spi.cpp line 65 

#include <U8glib-HAL.h> can't be found.

olikraus, 2022-01-03:
Hmm.. strange from where did you get the code?

olikraus, 2022-01-03:
Ah, I found the issue and fixed the same. Thanks for pointing out this issue.

olikraus, 2022-01-03:
I have taged the updated version in U8g2_Arduino with 2.32.7

closed 2022-01-07 by olikraus