olikraus/u8g2 #1750: Library with latest ESP32 Arduino does not work

Elektrik1, 2022-01-09:
Hi there,
U8g2 library stopped working with latest ESP32 Arduino version (2.0.1 and 2.0.2) due to changes to ESP libraries, so far I have found that it has something to do with multi-threaded I2C library changes, which does not satisfy current call to Wire library.

Basically code execution stops after u8g2.begin().

Not sure at which end it should be fixed, u8g2 library, or ESP32 Arduino library to maintain the compatibility.

olikraus, 2022-01-09:
U8g2 is developed against Arduino Wire Class. It U8g2 works with the original Arduino Wire implementation then probably it is a ESP32 issue if the Wire class had been reimplemented there.

me-no-dev, 2022-02-11:
This has little to do with RTOS itself. We had to refactor the Wire class to use ESP-IDF's driver interface, so that and the hardware design of the I2C bus on ESP32 made it necessary to support only strict Wire compatibility. Other I2C libs (OLEDs or not) function normally, so I would guess that there is some weird chain of Wire calls somewhere in this lib that is freaking the code.

v1993, 2022-02-11:
Ah, in my fork I don't use Wire class at all (or arduino compatibility module at all, for that matter), instead using C interface directly. It's also blazing fast on HW interfaces as a result and allows configuring HW pins.

Elektrik1, 2022-05-18:
I took the time to investigate, and found that I need additionally call the Wire.begin after u8g2 instance is called, and it should be called with SDA/SCL pins additionally. Like that:
```
#define SCREEN_SDA 16
#define SCREEN_SCL 17
U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, U8X8_PIN_NONE, SCREEN_SCL, SCREEN_SDA);
Wire.begin(SCREEN_SDA, SCREEN_SCL);
```

ESP32 Arduino versions <= 2.0.0 works fine without additional arguments to Wire.begin().

olikraus, 2022-05-18:
Hmm nice finding. Maybe I should use that extended syntax within u8g2

olikraus, 2022-08-13:
Looking at the u8g2 code it looks like this:
https://github.com/olikraus/u8g2/blob/d9402d5e40ca1224a300c39d9baef4051eefb7b5/cppsrc/U8x8lib.cpp#L1326-L1334

This means: If the constructor is 
`U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, U8X8_PIN_NONE, SCREEN_SCL, SCREEN_SDA);`
then
`Wire.begin(SCREEN_SDA, SCREEN_SCL);`
is executed in line 1329.
Doing the same command twice looks more like an issue in the ESP32 part than in u8g2.

closed 2022-08-13 by olikraus