olikraus/u8g2 #1754: Custom callback for u8g2_DrawStr

Sovichea, 2022-01-13:
Hi There,

Currently, I'm developing a text engine to support Khmer script in u8g2. In Khmer script, the order of typing is different from the order of text to be displayed. For example,

Typed: ស + ្ + រ
Display: ស្រ

For this reason, a text shaping algorithm must be implemented separately. For my case, I have the implementation of such algorithm, which can be found in my [Github page](https://github.com/Sovichea/u8g2-khmer-support). It works fine on its own until I start using `mui` library to create menu for my application. All functions in the library reference back to `u8g2_DrawStr` which call `u8g2_draw_string` function. 

I want to know if it is possible there is any workaround that I can make to switch between the default `u8g2_draw_string` and `u8g2_draw_string_khmer` whenever necessary.

Thank you,

olikraus, 2022-01-13:
1. Using MUI

Thanks for using MUI, however it is not yet released and very unstable. In fact MUI version from 2.31 is very outdated. There will be a lot of changes.

2. Callback for DrawStr

I personally would prefer to create a new function, which in turn calls DrawStr
Additionally I could imagine to create a callback just for MUI. Would this be ok?

Sovichea, 2022-01-13:
1. Yes I know that `mui` is not stable yet and I had to implement a lot of custom callback to suit my need. It working great in my application for now, but I also plan to update to code a long the way.

2. I think creating a callback to custom `DrawStr` would be a great solution. I imagine I can set up the font and draw string callback in a single different MUI_STYLE field, which will open up the support for other languages similar to Khmer script.

olikraus, 2022-01-13:
My idea was to keep font assignment and string two separate MUI commands (MUI_STYLE and MUI_LABEL for example)

https://github.com/olikraus/u8g2/blob/ef515e6c772ea9c1f6e67a993001ce11992845c6/sys/arduino/u8g2_page_buffer/MUIBlink/MUIBlink.ino#L479-L483

The MUIF definition of the corresponding style also got simplified meanwhile. From the same file:
https://github.com/olikraus/u8g2/blob/ef515e6c772ea9c1f6e67a993001ce11992845c6/sys/arduino/u8g2_page_buffer/MUIBlink/MUIBlink.ino#L378-L382

This means there will a new MUIF which allows you to set the font for the corresponding style number (no need to write your own callback for the style)

Having a callback for the draw callback might allow you to change the font also, but I think it will make the form description in MUI string less flexible.

olikraus, 2022-01-13:
The problem is more complicated. I would also need a callback for the u8g2_GetUTF8Width() width function.
Moreover I assume you mean the DrawUTF8 function instead of DrawStr

olikraus, 2022-01-14:
u8g2_GetUTF8Width() : Some MUI functions also require the width of the resulting string.This requires not only drawing the string but also the calculation of the width of the string. If you alter the drawing then probably you will also alter the width of the string.

Do you only require the callback for MUI_LABEL? If so, then the implementation would be simple and could be done already right now.

Sovichea, 2022-01-14:
> u8g2_GetUTF8Width() : Some MUI functions also require the width of the resulting string.This requires not only drawing the string but also the calculation of the width of the string

I see.

> Do you only require the callback for MUI_LABEL? If so, then the implementation would be simple and could be done already right now.

Actually, I have tried replacing the callback directly in `DrawUTF8` and I'm able to use it with other MUI functions to generate button and input fields with Khmer text. So, I'm thinking this implementation should support all MUI functions and not just MUI_LABEL.

olikraus, 2022-01-16:
Very nice. I added the callback topic to my list of todo's but I don't know when I will have time to work on this.

I have created beta release 2.32.8 including the latest MUI source. Please consider to use this for your menues. 
Manuals should be up to date (but still incomplete):
https://github.com/olikraus/u8g2/wiki/muiref
https://github.com/olikraus/u8g2/wiki/muimanual

MUI Example code is here:
https://github.com/olikraus/u8g2/tree/master/sys/arduino/u8g2_page_buffer/MUIBlink
https://github.com/olikraus/u8g2/tree/master/sys/arduino/u8g2_page_buffer/MUICountDown
https://github.com/olikraus/u8g2/tree/master/sys/arduino/u8g2_page_buffer/MUIStopwatch
...

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

Sovichea, 2022-01-17:
That's great! Thank you.

closed 2022-01-17 by Sovichea