olikraus/u8g2 #1755: Speed up U8g2 with ST7920

AntoRoni, 2022-01-16:
Is there any way to speed up the st7920 display refresh time, I did the tests on my esp32 and it currently takes ~110ms to do a sendBuffer()

olikraus, 2022-01-16:
There are always ways to speed up procedure. I try to speed up u8g2 as much as possible, but the primary goal is reliability and portability. This will not allow some optimizations.
I think in your case the Arduino procedures are the limiting factors.

If you refer to HW SPI, my only suggestion is to change the SPI procedures by yourself.

https://github.com/olikraus/u8g2/blob/e3c8a4ae005bee74ae45fcbdab0e60dbe698d65f/cppsrc/U8x8lib.cpp#L865-L986

AntoRoni, 2022-01-16:
I'm using SW SPI

olikraus, 2022-01-16:
Same answer... you need to modify the code.
Maybe reduce the us value here:
https://github.com/olikraus/u8g2/blob/e3c8a4ae005bee74ae45fcbdab0e60dbe698d65f/cppsrc/U8x8lib.cpp#L545

AntoRoni, 2022-01-16:
I'm using 3 wire SW SPI for ST7920, what do you suggest to do in the code in that case?

`extern "C" uint8_t u8x8_byte_arduino_3wire_sw_spi(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
  uint8_t i;
  uint8_t takeover_edge = u8x8_GetSPIClockPhase(u8x8);
  uint16_t b;
  uint8_t *data;

  /* the following static vars are recalculated in U8X8_MSG_BYTE_START_TRANSFER */
  /* so, it should be possible to use multiple displays with different pins */
  
  static volatile uint8_t *arduino_clock_port;
  
  static uint8_t arduino_clock_mask;
  static uint8_t arduino_clock_n_mask;
  
  static volatile uint8_t *arduino_data_port;
  static uint8_t arduino_data_mask;
  static uint8_t arduino_data_n_mask;

  static uint8_t last_dc;

  switch(msg)
  {
    case U8X8_MSG_BYTE_SEND:
    
      data = (uint8_t *)arg_ptr;      
      if ( takeover_edge == 0 )
      {
	while( arg_int > 0 )
	{
	  b = *data;
	  if ( last_dc != 0 )
	    b |= 256;
	  data++;
	  arg_int--;
	  /* issue 156, check for speed */
#if F_CPU <= 17000000
	  if ( b == 0 )
	  {
	    *arduino_data_port &= arduino_data_n_mask;
	    for( i = 0; i < 9; i++ )
	    {
	      *arduino_clock_port |= arduino_clock_mask;	    
	      *arduino_clock_port &= arduino_clock_n_mask;
	    }
	  }
	  else
#endif
	  {
	    for( i = 0; i < 9; i++ )
	    {
	      if ( b & 256 )
		*arduino_data_port |= arduino_data_mask;
	      else
		*arduino_data_port &= arduino_data_n_mask;

	      *arduino_clock_port |= arduino_clock_mask;	    
	      b <<= 1;
	      *arduino_clock_port &= arduino_clock_n_mask;
	    }
	  }
	}
      }
      else
      {
	while( arg_int > 0 )
	{
	  b = *data;
	  if ( last_dc != 0 )
	    b |= 256;
	  data++;
	  arg_int--;
	  /* issue 156, check for speed */
#if F_CPU <= 17000000
	  if ( b == 0 )
	  {
	    *arduino_data_port &= arduino_data_n_mask;
	    for( i = 0; i < 9; i++ )
	    {
	      *arduino_clock_port &= arduino_clock_n_mask;
	      *arduino_clock_port |= arduino_clock_mask;	    
	    }
	  }
	  else
#endif
	  {
	    for( i = 0; i < 9; i++ )
	    {
	      if ( b & 256 )
		*arduino_data_port |= arduino_data_mask;
	      else
		*arduino_data_port &= arduino_data_n_mask;

	      *arduino_clock_port &= arduino_clock_n_mask;
	      b <<= 1;
	      *arduino_clock_port |= arduino_clock_mask;	    
	    }
	  }
	}
      }      
      break;
      
    case U8X8_MSG_BYTE_INIT:
      /* disable chipselect */
      u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_disable_level);
      /* no wait required here */
      
      /* for SPI: setup correct level of the clock signal */
      u8x8_gpio_SetSPIClock(u8x8, u8x8_GetSPIClockPhase(u8x8));
      break;
    case U8X8_MSG_BYTE_SET_DC:
      last_dc = arg_int;
      break;
    case U8X8_MSG_BYTE_START_TRANSFER:
      u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_enable_level);  
      u8x8->gpio_and_delay_cb(u8x8, U8X8_MSG_DELAY_NANO, u8x8->display_info->post_chip_enable_wait_ns, NULL);

      /* there is no consistency checking for u8x8->pins[U8X8_PIN_SPI_CLOCK] */
    
      arduino_clock_port = portOutputRegister(digitalPinToPort(u8x8->pins[U8X8_PIN_SPI_CLOCK]));
      arduino_clock_mask = digitalPinToBitMask(u8x8->pins[U8X8_PIN_SPI_CLOCK]);
      arduino_clock_n_mask = ~arduino_clock_mask;
    
      /* there is no consistency checking for u8x8->pins[U8X8_PIN_SPI_DATA] */

      arduino_data_port = portOutputRegister(digitalPinToPort(u8x8->pins[U8X8_PIN_SPI_DATA]));
      arduino_data_mask = digitalPinToBitMask(u8x8->pins[U8X8_PIN_SPI_DATA]);
      arduino_data_n_mask = ~arduino_data_mask;
      
      break;
    case U8X8_MSG_BYTE_END_TRANSFER:
      u8x8->gpio_and_delay_cb(u8x8, U8X8_MSG_DELAY_NANO, u8x8->display_info->pre_chip_disable_wait_ns, NULL);
      u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_disable_level);
      break;
    default:
      return 0;
  }
  return 1;
}

#else
  /* fallback */
  uint8_t u8x8_byte_arduino_3wire_sw_spi(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
  {
    return u8x8_byte_3wire_sw_spi(u8x8, msg,arg_int, arg_ptr);
  }
  
#endif`

olikraus, 2022-01-16:
3-wire SPI will not work with ST7920. The ST7920 only supports 4-wire SPI (although you probably use 3 wires :-D )

AntoRoni, 2022-01-16:
so why in the examples it is like this:
`U8G2_ST7920_128X64_F_SW_SPI u8g2(U8G2_R0, /* clock=*/ 18 /* A4 */ , /* data=*/ 16 /* A2 */, /* CS=*/ 17 /* A3 */, /* reset=*/ U8X8_PIN_NONE);`

olikraus, 2022-01-16:
It will call "u8x8_byte_arduino_4wire_sw_spi".

AntoRoni, 2022-01-16:
Thank you I managed to decrease the time from ~110ms to ~30ms by going to "**u8x8_byte_4wire_sw_spi**" and replacing the "**u8x8_gpio_Delay**" with several `__asm__ __volatile__ ("nop");`

AntoRoni, 2022-01-16:
where is the code for these functions defined: "**u8x8->gpio_and_delay_cb**" , "**u8x8_gpio_SetSPIClock**" ?

olikraus, 2022-01-16:
"u8x8->gpio_and_delay_cb" is a function pointer and the function is probably this:
https://github.com/olikraus/u8g2/blob/e3c8a4ae005bee74ae45fcbdab0e60dbe698d65f/cppsrc/U8x8lib.cpp#L83

"u8x8_gpio_SetSPIClock" :
https://github.com/olikraus/u8g2/blob/e3c8a4ae005bee74ae45fcbdab0e60dbe698d65f/csrc/u8x8.h#L778

AntoRoni, 2022-01-16:
Thank you. I made several twaks and now the refresh rate is very fast.

closed 2022-01-16 by AntoRoni