olikraus/u8g2 #1758: Hardware SPI not working with Adafruit RP2040

tkreyche, 2022-01-18:
I'm looking for suggestions on what to try next to get hardware SPI working using the Adafruit RP2040 board to drive a DOGXL240. Software SPI work fine. The RP2040 is running the Raspberry Pi Pico Arduino core by earlephilhower [https://github.com/earlephilhower/arduino-pico] The SPI pin mapping looks correct and for both SW and HW I'm using the same pins.

#define PIN_SPI0_MISO  (16u)
#define PIN_SPI0_MOSI  (19u)
#define PIN_SPI0_SCK   (18u)

// SW works great
U8G2_UC1611_EA_DOGXL240_F_4W_SW_SPI u8g2(U8G2_R2, /* clock=*/ 18, /* data=*/ 19, /* cs=*/ 1, /* dc=*/ 6, /* reset=*/ 0);
//HW SPI not working
//U8G2_UC1611_EA_DOGXL240_F_4W_HW_SPI u8g2(U8G2_R2, /* cs=*/ 1, /* dc=*/ 6, /* reset=*/ 0);

SPI bus traces show the HW version not even close to the SW version. See attachments for traces, hardware config and sample code.

This library is an awesome achievement! 

thanks, Tom

[image]
[code.txt](https://github.com/olikraus/u8g2/files/7892567/code.txt)
[image]
[image]

tkreyche, 2022-01-19:
Tried the same test with the same hardware and got the same results using the Arduino Core for mbed instead of the earlephilhower core. Pin mapping supposed to be standard but I must be missing something.

tkreyche, 2022-01-19:
Got HW SPI working with the Arduino Core for mbed by changing the pin assignment in:

..\AppData\Local\Arduino15\packages\arduino\hardware\mbed_rp2040\2.6.1\variants\RASPBERRY_PI_PICO\pins_arduino.h

It's much faster than software SPI. Note that there is some discussion on forums that changing pin assignments in the .h file is not the preferred approach, but I don't see any choice since the U8g2 hides the pins.

//#define PIN_SPI_MOSI  (3u)
//#define PIN_SPI_SCK   (2u)
#define PIN_SPI_MOSI  (19u)
#define PIN_SPI_SCK   (18u)

tkreyche, 2022-01-19:
Demo program using DOGXL ran fine overnight using the RP2040 with Arduino Core for mbed. I'll let someone else figure out the earlephilhower core.

closed 2022-01-19 by tkreyche

olikraus, 2022-01-19:
> but I don't see any choice since the U8g2 hides the pins.

U8g2 does not hide anything. In fact U8g2 does not know or require the SPI pins (MISO, MOSI, SCK). U8g2 will just use SPI.h from Arduino lib. As a consequence, you need to connect to the pins, which are used by SPI.h library.
Even with the SPI.h library you can not assitgn pin. And also SPI.h does not know the SPI pins.

But I am happy that you figured out a way to make it work.

tkreyche, 2022-01-19:
Thanks for the correction and great library!