olikraus/u8g2 #1759: Display dimension 122X32 with ST7567 Driver not listed

amir-skn, 2022-01-19:
Thank you for your great Library!

My lcd has two 122X16 sections above each other which makes a 122X32 display with a ST7567 driver. I really hoped that it was listed in supported configuration but it is not. Triple checked.  Data Sheet is attached
[SDGG12232-07.pdf](https://github.com/olikraus/u8g2/files/7894362/SDGG12232-07.pdf)

olikraus, 2022-01-19:
I never in my life had a problem where I found a ready made solution which 100% fits. I am sure there are thousands of displays out there. Do you really expect we do buy all displays and add support for it?

Try one or more of the ST7567 constructors, for example:
U8G2_ST7567_JLX12864_1_4W_SW_SPI(rotation, clock, data, cs, dc [, reset]) [page buffer, size = 128 bytes]

Of course: dimensions will not fit and contrast will be wrong. Maybe the picture is rotated or some how destroyed.
But we need to go ahead and fix one issue after the other. I can support, but the testing has to be done by you.
First step: Make something visible. Some pixel should become dark.
Use the constructor from above, use the setContrast to adjust the contrast. maybe also use the contrast test (https://github.com/olikraus/u8g2/tree/master/sys/arduino/u8g2_page_buffer/ContrastTest)
Will you see something? If yes, send a picture here.

amir-skn, 2022-01-23:
You are 100% right! thank you for your support. Sorry for the delay: I started to use the constructor that you've mentioned and got stuck in the very beginning. I was using the ESP-IDF extension in VSCode with "Arduino as a component" added to it. whenever the code reached the `begin()`, the esp panicked. I traced it down to a memset function , it was accessing a wrong area of the memory. I've found ESP-IDF + Arduino-as-a-component in a bleeding-edge state and not a stable enough approach. So I switched to PlatformIO:

I ran the following code (Clock Example, contrast =7) successfully.
here is the code :
`

#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

U8G2_ST7567_JLX12864_1_4W_SW_SPI u8g2(U8G2_R2, /* clock=*/18, /* data=*/23, /* cs=*/19, /* dc=*/22, /* reset=*/5);

void setup(void)
{

  u8g2.begin();
  u8g2.setContrast(7);
}

uint8_t m = 24;

void loop(void)
{
  char m_str[3];
  strcpy(m_str, u8x8_u8toa(m, 2)); /* convert m to a string with two digits */
  u8g2.firstPage();
  do
  {
    u8g2.setFont(u8g2_font_logisoso62_tn);
    u8g2.drawStr(0, 63, "9");
    u8g2.drawStr(33, 63, ":");
    u8g2.drawStr(50, 63, m_str);
  } while (u8g2.nextPage());
  delay(1000);
  m++;
  if (m == 60)
    m = 0;
}`

and the result:
[image]

olikraus, 2022-01-23:
Very nice, good picture. Looks like it almost works.
So this is a 122x32 display with a 3 pixel gap between two ares of 122x16 pixel.
I will try to create a 122x32 constructor based on your picture and code.

amir-skn, 2022-01-23:
I don't wether it would help or not. but I've changed the `u8x8_st7567_jlx12864_display_info`  in **u8x8_d_st7567.c**
to this:

```
static const u8x8_display_info_t u8x8_st7567_jlx12864_display_info =
{
  /* chip_enable_level = */ 0,
  /* chip_disable_level = */ 1,
  
  /* post_chip_enable_wait_ns = */ 150,	/* */
  /* pre_chip_disable_wait_ns = */ 50,	/* */
  /* reset_pulse_width_ms = */ 1, 
  /* post_reset_wait_ms = */ 1, 
  /* sda_setup_time_ns = */ 50,		/* */
  /* sck_pulse_width_ns = */ 120,	/* */
  /* sck_clock_hz = */ 4000000UL,	/* */
  /* spi_mode = */ 0,		/* active high, rising edge */
  /* i2c_bus_clock_100kHz = */ 4,
  /* data_setup_time_ns = */ 40,	/* */
  /* write_pulse_width_ns = */ 80,	/* */
  /* tile_width = */ 16,		/* width of 16*8=128 pixel */
  /* tile_hight = */ 4,
  /* default_x_offset = */ 4,
  /* flipmode_x_offset = */ 0,
  /* pixel_width = */ 128,
  /* pixel_height = */ 32
};
```

And flashed this piece of code:

```
#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

U8G2_ST7567_JLX12864_F_4W_SW_SPI u8g2(U8G2_R2, /* clock=*/15, /* data=*/13, /* cs=*/23, /* dc=*/2, /* reset=*/U8X8_PIN_NONE);

void setup(void)
{

  u8g2.begin();
  u8g2.setContrast(7);
}

uint8_t m = 24;

void loop(void)
{
  char m_str[3];
  strcpy(m_str, u8x8_u8toa(m, 2)); /* convert m to a string with two digits */
  u8g2.firstPage();
  do
  {
    u8g2.setFont(u8g2_font_pxplustandynewtv_8_all);
    u8g2.drawStr(0, 8, "9");
    u8g2.drawStr(63, 16, ":");
    u8g2.drawStr(118, 32, m_str);
  } while (u8g2.nextPage());
  delay(1000);
  m++;
  if (m == 60)
    m = 0;
}
```

the result is like this:
[image]

olikraus, 2022-01-23:
I already created a new constructor: U8G2_ST7567_122X32_1_4W_SW_SPI (and friends).
There are several topics and indeed we need to do some reverse engineering.

Bias 1/6
Your datasheet advices us to use 1/6 Bias. I changed this in the code, but this will probably affect the contrast. As a result your contrast(7) from your second post will most probably be wrong. So we need to figure out the new contrast value

Dimensions
I have adjusted the dimensions of the new constructor according to your image, but I do not know, whether everything is correct.
Please use the new code and run the page mode "FlipMode.ino" example (https://github.com/olikraus/u8g2/tree/master/sys/arduino/u8g2_page_buffer/FlipMode)
This example will cycle through several modes and it will always draw a frame next to the border of the display.
If the frame is there all the time, then everything is ok.

I have created new u8g2 version 2.32.10

I do not use platform io, here are the instructions for Arduino IDE:

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

Can you check contrast and display dimensions with the new code and constructor?

olikraus, 2022-01-23:
> If the frame is there all the time, then everything is ok.

If not: A picture of that display content would be great.

amir-skn, 2022-01-23:
> I do not use platform io, here are the instructions for Arduino IDE:
We just need to add this lines to `platformio.ini` : 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```
 
I didn't set contrast at all and EVERYTHING IS WORKING GREAT! THANKS!
[image]

closed 2022-01-23 by amir-skn

olikraus, 2022-01-23:
Nice, thanks for testing.