olikraus/u8g2 #1760: String issue

pike2k, 2022-01-22:
I'm using the latest version for PIO (nice that it have updated) 2.28.8

When I combine strings like this:

  String temperature = String(sensors.getTempC(sensor1),1)+"°C";
  front.print(temperature);

The resulting output on display becomes "21.5B°C" I have no idea where this B comes from. It is not in the temperature variable if I serial output it.

pike2k, 2022-01-22:
The only thing that might? be odd with my setup is that I'm using 2 Oled displays at the same time, one over HW_I2C and another over HW_SPI. One is front. and the other is rear. 
Both shows the same issue.

pike2k, 2022-01-23:
I changed place between '°' and 'C' (so C°) and the output then became "CB°" So seems like '°' is the troublemaker. 
Am I doing something wrong?

v1993, 2022-01-23:
Do you call [`u8g2.enableUTF8Print();`](https://github.com/olikraus/u8g2/wiki/u8g2reference#enableutf8print) as part of setup? If not, you should.

pike2k, 2022-01-23:
Thanks. That does the trick. Is this a recent requirement?

v1993, 2022-01-23:
Pretty sure this was required since introduction of UTF8 support to u8g2's print function.

closed 2022-01-23 by pike2k