olikraus/u8g2 #1764: sh1122 128x64 oled and  esp8266

teeshq, 2022-01-24:
hi im quite noob how to use sh1122 128x64 and esp8266 nodemcu how to define spi pins

olikraus, 2022-01-24:
If I remember correctly, you could append the SPI signals (ESP pin remapping feature):

`U8G2_SH1122_256X64_1_4W_HW_SPI(rotation, cs, dc, reset, clock, data) `

For testing you could also use the SW SPI constructor

`U8G2_SH1122_256X64_1_4W_SW_SPI(rotation, clock, data, cs, dc, reset) `

teeshq, 2022-01-26:
> If I remember correctly, you could append the SPI signals (ESP pin remapping feature):
> 
> `U8G2_SH1122_256X64_1_4W_HW_SPI(rotation, cs, dc, reset, clock, data) `
> 
> For testing you could also use the SW SPI constructor
> 
> `U8G2_SH1122_256X64_1_4W_SW_SPI(rotation, clock, data, cs, dc, reset) `
 hi thanks a lot `U8G2_SH1122_256X64_1_4W_SW_SPI(rotation, clock, data, cs, dc, reset) ` work 
have another question its look like oled have much slower refresh rate then mega 2560
in same code on mega2560 `u8g2.setBusClock(2000000)` and look great but in ESP look slow
thanks again

olikraus, 2022-01-27:
You could try `U8G2_SH1122_256X64_1_4W_HW_SPI(rotation, cs, dc, reset, clock, data)` 
(clock and data are moved to the end of the argument list)

teeshq, 2022-01-27:
> You could try `U8G2_SH1122_256X64_1_4W_HW_SPI(rotation, cs, dc, reset, clock, data)` (clock and data are moved to the end of the argument list)<
`GraphicsTest:55:117: error: no matching function for call to 'U8G2_SH1122_256X64_F_4W_HW_SPI::U8G2_SH1122_256X64_F_4W_HW_SPI(const u8g2_cb_t*, int, int, int, int, int)'
   55 |   U8G2_SH1122_256X64_F_4W_HW_SPI u8g2(U8G2_R2, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8, /* clock=*/ 14, /* data=*/ 13);    // Enable U8G2_16BIT in u8g2.h
      |                                                                                                                     ^
In file included from C:\Users\teesh\Documents\Arduino\GraphicsTest\GraphicsTest.ino:37:
C:\Users\teesh\Documents\Arduino\libraries\U8g2\src/U8g2lib.h:3969:11: note: candidate: 'U8G2_SH1122_256X64_F_4W_HW_SPI::U8G2_SH1122_256X64_F_4W_HW_SPI(const u8g2_cb_t*, uint8_t, uint8_t, uint8_t)'
 3969 |   public: U8G2_SH1122_256X64_F_4W_HW_SPI(const u8g2_cb_t *rotation, uint8_t cs, uint8_t dc, uint8_t reset = U8X8_PIN_NONE) : U8G2() {
      |           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\teesh\Documents\Arduino\libraries\U8g2\src/U8g2lib.h:3969:11: note:   candidate expects 4 arguments, 6 provided
C:\Users\teesh\Documents\Arduino\libraries\U8g2\src/U8g2lib.h:3968:7: note: candidate: 'constexpr U8G2_SH1122_256X64_F_4W_HW_SPI::U8G2_SH1122_256X64_F_4W_HW_SPI(const U8G2_SH1122_256X64_F_4W_HW_SPI&)'
 3968 | class U8G2_SH1122_256X64_F_4W_HW_SPI : public U8G2 {
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\teesh\Documents\Arduino\libraries\U8g2\src/U8g2lib.h:3968:7: note:   candidate expects 1 argument, 6 provided
C:\Users\teesh\Documents\Arduino\libraries\U8g2\src/U8g2lib.h:3968:7: note: candidate: 'constexpr U8G2_SH1122_256X64_F_4W_HW_SPI::U8G2_SH1122_256X64_F_4W_HW_SPI(U8G2_SH1122_256X64_F_4W_HW_SPI&&)'
C:\Users\teesh\Documents\Arduino\libraries\U8g2\src/U8g2lib.h:3968:7: note:   candidate expects 1 argument, 6 provided
exit status 1
no matching function for call to 'U8G2_SH1122_256X64_F_4W_HW_SPI::U8G2_SH1122_256X64_F_4W_HW_SPI(const u8g2_cb_t*, int, int, int, int, int)'
`
cant get it to work

teeshq, 2022-01-29:
Sorted

closed 2022-01-29 by teeshq