olikraus/u8g2 #1765: Color inversion (ST7565 / EA DOGM132)

mws-rmain, 2022-01-27:
What is the best way to implement color inversion, which is supported by the ST7565)?
U8G2 code appears to always set 'normal' color mapping, and does not expose a method to toggle it.
For instance, in u8x8_d_st7565.c:1235:
`U8X8_C(0x0a6),		                /* display normal, bit val 0: LCD pixel off. */
`
The low bit of this command (0x0A6 / 0x0A7) will 'reverse' the color mapping of the display (dark pixels are light, & vice-versa).
[image]

We normally use the (transflective) EA DOGM132W-5 display, which currently has availability issues. The (non-transflective) EA DOGM132S-5 is available. Unfortunately, when driven with the same code, it defaults to 'color-inverted', compared with the 132W.
I'd like a quick & easy way to allow the user to toggle this setting.

I'd like to extend the u8x8_d_st7565_common() function to handle 'color inversion' with a new message, similar to:
- U8X8_MSG_DISPLAY_SET_POWER_SAVE; or
- U8X8_MSG_DISPLAY_SET_CONTRAST.

I'd like your thoughts on how best to implement this - I have no idea if this 'feature' is peculiar to the ST7565, or has broader application.

Thanks,
Robb.

olikraus, 2022-01-27:
There is a sendF command to do those low level controller specific things.
You could also draw a full screen box in color 1 and the draw the rest of your graphics commands in color 0.

olikraus, 2022-01-27:
https://github.com/olikraus/u8g2/wiki/u8g2reference#sendf

The example for the sendf command also answers your question.

mws-rmain, 2022-01-27:
Thanks, Oli.
I'll give sendF a try.
Cheers,
Robb.

mws-rmain, 2022-01-28:
Thanks again, Oli.
The example you pointed to for sendF works perfectly.
Cheers,
Robb.

closed 2022-01-28 by mws-rmain