olikraus/u8g2 #1768: Support for display MT-12864K RW1065I

igor605ds, 2022-01-31:
Hi, olikraus!

Thanks for the excellent U8g2 library.

I need your help, U8g2 library does not support MT-12864K display
https://amperka.ru/product/display-lcd-graphic-128x64-i2c-fla

Display documentation:
http://wiki.amperka.ru/_media/products:display-lcd-graphic-128x64-i2c:mt-12864k-datasheet.pdf
 
I have a test program for working with a display.
https://gist.github.com/igor605ds/a6368c8f48f762fbb18eadf52ca65ff9

However, I would like to make this display compatible with the U8g2 library. How can I do that?
[image]

olikraus, 2022-01-31:
Adding support for this display will be VERY complicated. At least I do not have the time to do this at the moment. Maybe you can follow the discussion here https://github.com/olikraus/u8g2/discussions/1767 and try this by your own.

igor605ds, 2022-02-01:
I'm ready to add display support myself. Help then, what I need to follow the procedure for adding a new driver.
Especially, as I understand it, some files are automatically generated using:
https://github.com/olikraus/u8g2/blob/master/tools/codebuild/codebuild.c

igor605ds, 2022-02-02:
Hi, olikraus!

The MT-12864K display does not have a multiplexer.

There are three chips on the display board:
- RW1065I driver control of the left half of the display (0x3C)
- RW1065I driver control of the right half of the display (0x3D)
- INF8475A (PCF8574A) additional chip for software control of backlight and contrast (0x3B). The INF8475A chip is not used to communicate with RW1065I controllers.

https://skr.sh/sCJZqMljWyE

The display manufacturers claim that the commands of the RW1065I chip are compatible with the KS0108.

The main differences between the chips:

- KS0108 parallel interface 8-bit.
- KS0108 left or right crystal selection by CS0 and CS1.

- RW1065I I²C interface
- RW1065I select left or right chip with address 0x3C or 0x3D.

The Arduino I²C scanner always gives three addresses on the I²C bus:
https://skr.sh/sCJE2c19zEa

And could you answer the question of whether this file should be used to generate library files:
https://github.com/olikraus/u8g2/blob/master/tools/codebuild/codebuild.c

olikraus, 2022-02-02:
> The MT-12864K display does not have a multiplexer.

Great, then everything will be a little bit easier.

> And could you answer the question of whether this file should be used to generate library files:
https://github.com/olikraus/u8g2/blob/master/tools/codebuild/codebuild.c

yes, you need a similar entry like this:

https://github.com/olikraus/u8g2/blob/ef9fa9f8f8aaa7767c7dd959dcf512433ddc2047/tools/codebuild/codebuild.c#L1510-L1517

However you will use COM_I2C and maybe we need a replacement for u8x8_cad_001.
Maybe "u8x8_cad_ssd13xx_i2c" fits, but I didn't studied the i2c protocol of the RW1065.

igor605ds, 2022-02-04:
Hi, olikraus!

Thanks for your help.

1. I added the controller to the `codebuild.c` file:
https://github.com/igor605ds/u8g2/commit/a111fc994d444b62c20ffc355ccf0dd52363886b

2. Then rebuild the code in file `codebuild.c`. As a result, I received an update of the files:
https://github.com/igor605ds/u8g2/commit/1d38325d932a3d561859ab1ab56d7a1b1a0e66c1

Question:
I now need to create an empty file `u8x8_d_rw1065i.c`? And then write the initialization of the display in it?

olikraus, 2022-02-04:
Correct, I quote myself:

> I think the best way is to start with the ks0108 driver for 128x64. It looks like that your display is somehow compatible with the KS0108.

So you could use the u8x8_d_ks0108.c as a template for the new file.

closed 2022-03-04 by olikraus

igor605ds, 2022-03-04:
Hi, Olikraus!

I am now on the territory of Ukraine and we have a war.

When this is over, if you don't mind, I'll get back to the display problem.

olikraus, 2022-03-04:
oh... then please reopen this issue whenever you have time.

Take care

nickgagin, 2022-12-05:
Can I offer help to @igor605ds on this topic, and continue a series of questions here? Or should I open new issue?

nickgagin, 2022-12-05:
> Depends on the questions... well, usually I prefer a new issue...

Questions about adding of support for MT-12864K with INF8475A (PCF8574A) and RW1065I i2c controllers onboard )
Probably, "Discussion" is more suitable for topic then "Issue".

1. I added `csrc/u8x8_d_rw1065i_mt12864k.c`, almost empty by your suggestion. 
2. Copied structure `u8x8_display_info_t` from ssd1306 for reference as u8x8_display_info_t u8x8_rw1065i_mt12864k_display_info ```
3. Added to codebuild.c this: `/* KS0108-like I2C interface */
  {
    /* name */ "rw1065i",
    /* tile_width */ 24,
    /* tile_height */ 8,
    /* ll_hvline */ "u8g2_ll_hvline_vertical_top_lsb",
    /* cad */ "u8x8_cad_001",
    /* cad_shortname */ "",
    /* com */ COM_KS0108,
    /* note */ "",
    /* is_generate_u8g2_class= */ 1,
    /* display_list */
    {
      { "128x64" },
      { NULL }
    },
  },`

I would like to clarify what is meant by the tile size? Addressable display memory page according to datasheet of IC?

olikraus, 2022-12-05:
One tile is a 8x8 pixel area (64 pixel). U8g2 usually transfers one tile as smallest quantity.

nickgagin, 2022-12-06:
@olikraus,  correct me if I wrong, but all manipulations with i2c address of controller of display I should implement in u8x8_cad.c?

If so, is there any possibility to redefine I2C addresses of controllers through u8g2 constructors?
By the way, I look over all source code and didn't find any realization for displays with multiple controllers, am I right? May be I missed something?