olikraus/u8g2 #1769: U8G2 7 segment font

Oregos29, 2022-01-31:
Hi,
I was looking for a 7 segments font usable with u8g2 librairie to print numbers.
I did’nt find exactly what I am looking for (hight and thin font), so I create it with fony and the bdfconv converter (thanks Oliver for all).

This is two fonts with :
Space + - . , (/ use for space as large as point) 0 1 2 3 4 5 6 7 8 9 :

Size : 
33x19
and
41x21

May be that will help somebody.
Regards

olikraus, 2022-01-31:
Nice, thanks a lot. Is it ok to add your fonts to u8g2?

Oregos29, 2022-01-31:
Yes, it is ok.

olikraus, 2022-02-08:
added to font build

Oregos29, 2022-03-20:
Hi,
I have create a new font for a clock project using a 8x32 led module.
There is number and letters.
[3x7_7_seg_Font.zip](https://github.com/olikraus/u8g2/files/8310417/3x7_7_seg_Font.zip)

olikraus, 2022-04-03:
I make the above font available in this issue here.

[image]

Just copy and paste the below array in your code and use `u8g2_font_3x7_multi` as argument for the setFont command.

```
/*
  Fontname: 3x7_Multi_v7
  Copyright: Created with Fony 1.4.0.2
  Glyphs: 97/256
  BBX Build Mode: 2
*/
const uint8_t u8g2_font_3x7_multi[1151] U8G2_FONT_SECTION("u8g2_font_3x7_multi") = 
  "a\2\5\2\4\4\1\1\5\10\11\0\0\11\2\11\2\1K\2\361\4b \6\230\343O\12!\11\230"
  "c\220\323\275\134\0\42\6\230\343O\12#\17\230\343\224\20\62 \222\20\62 \222\20\16$\6\230\343O"
  "\12%\16\230c\220\220\23#\23#\223\223\220\12&\6\230\343O\12'\6\230\343O\12(\13\230\343\20"
  "#\223SG'\26)\13\230c\220\243\323\214L.\0*\6\230\343O\12+\11\230\343\30\63\223\217\3"
  ",\7\230\343\217\230\13-\7\230c\274\17\7.\7\230\343\217\230\13/\14\230c\221\23#\23#\223\223"
  "\13\60\21\230c\260\222\220\222\220\222\220\222\220\222\220\262\12\61\12\230\343\20\243\323\231U\0\62\13\230c"
  "\260Se%'g\25\63\13\230c\260Se\247\312*\0\64\16\230c\220\220\222\220\222\220\262\323*\0"
  "\65\13\230c\260\222\223\263Se\25\66\16\230c\260\222\223\263\222\220\222\220\262\12\67\10\230c\260\323W"
  "\1\70\20\230c\260\222\220\222\220\262\222\220\222\220\262\12\71\15\230c\260\222\220\222\220\262Se\25:\10"
  "\230c\230\227\17\4;\6\230\343O\12<\11\230c\25S\250:\0=\10\230c\270\266\17\3>\10\230"
  "c\324L}\0\77\14\230c\260\222\220\23#\223\27\13@\21\230cp !%!a\241/,\324X"
  "X\5A\21\230c\260\222\220\222\220\262\222\220\222\220\222\220\12B\20\230c \223\220\222\220\262\222\220\222"
  "\220\42\13C\12\230c\260\222\323\235U\0D\21\230c \223\220\222\220\222\220\222\220\222\220\42\13E\14"
  "\230c\260\222\223#\223\223\263\12F\13\230c\260\222\223#\223S\27G\16\230c\260\222S!%!%"
  "!e\25H\22\230c\220\220\222\220\222\220\262\222\220\222\220\222\220\12I\12\230c\60\223\323\231U\0J"
  "\15\230c\60\223\223\243\223\222\220\262\12K\21\230c\220\223\220\222\220\42\223\220\222\220\222\220\12L\11\230"
  "c\220\323;\253\0M\22\230c\220\220\262\222\220\222\220\222\220\222\220\222\220\12N\15\230c\230\263\222\220"
  "\222\220\222\220\12O\21\230c\260\222\220\222\220\222\220\222\220\222\220\262\12P\15\230c\260\222\220\222\220\262"
  "\222S\27Q\21\230c\260\222\220\222\220\222\220\222\220\42\223\220\12R\20\230c\260\222\220\222\220\262\42\223"
  "\220\222\220\12S\13\230c\260\222\223\263Se\25T\11\230c\60\223\323\263\0U\22\230c\220\220\222\220"
  "\222\220\222\220\222\220\222\220\262\12V\16\230c\220\220\222\220\222\220r&'\26W\22\230c\220\220\222\220"
  "\222\220\222\220\222\220\262\222\220\12X\20\230c\220\220\222\220\62\23\263\222\220\222\220\12Y\16\230c\220\220"
  "\222\220\222\220\62\223S\26Z\14\230c\260\23#\23#\223\263\12[\12\230c \223\323\35Y\0\134\13"
  "\230c\220\223\243\223\243S\25]\11\230c\240\323\63\262\0^\11\230\343\20\223\220\17\17_\7\230\343\217"
  "\270\12`\6\230\343O\12a\14\230c\270\222\220\222\220\262\223\12b\17\230c\220\223\263\222\220\222\220\222"
  "\220\262\12c\11\230c\270\222Sg\25d\17\230c\221\223\262\222\220\222\220\222\220\262\12e\13\230c\270"
  "\222\220\262\222\263\12f\13\230\343\240\222\223#\223S\27g\13\230c\270\222\220\262\223\262\12h\20\230c"
  "\220\223\263\222\220\222\220\222\220\222\220\12i\11\230\343\230\227\23\263\12j\13\230\343\220\227\223\243\223\262\12"
  "k\17\230c\224\223\220\42\223\220\222\220\222\220\12l\10\230\343\224\323\263\0m\21\230c\224\63\240\221P"
  "#\241FB\215\204\232\0n\17\230c\224\263\222\220\222\220\222\220\222\220\12o\15\230c\270\222\220\222\220"
  "\222\220\262\12p\13\230c\270\222\220\262\222\223\13q\12\230c\270\222\220\262S\25r\11\230\343\250\222\323"
  ".\0s\12\230c\270\222\263\223\262\12t\13\230c\220\223#\223Sg\25u\16\230c\230\220\222\220\222"
  "\220\222\220\262\12v\15\230c\230\220\222\220\222\220\62\23\13w\17\230c\230Q\241FB\215\1\221\204P"
  "\0x\15\230c\230\220\222\20S!%!\25y\14\230c\230\220\222\220\262\223\262\12z\11\230c\270\23"
  "Sg\25{\13\230\343 \223\23\243SG\25|\10\230c\220\323w\1}\13\230c\240SG&'F"
  "\26~\12\230c\270\222Pe\37\4\177\6\230\343O\12\200\6\230\343O\12\0\0\0\4\377\377\0";
```

closed 2022-04-03 by olikraus