olikraus/u8g2 #1771: Sharp ls013b7dh05_144x168 uses SPI mode 4 in HW mode, should use 0

davehummel, 2022-02-03:
The current SPI initialization leaves a banded screen with artifacts.  This code corrects the problem locally (tested on Adafruit display with teensy 3.6), but should be applied to the library - 

`static const u8x8_display_info_t dh_u8x8_ls013b7dh05_144x168_display_info =
    {
        /* chip_enable_level = */ 1,
        /* chip_disable_level = */ 0,
        /* post_chip_enable_wait_ns = */ 50,
        /* pre_chip_disable_wait_ns = */ 50,
        /* reset_pulse_width_ms = */ 1,
        /* post_reset_wait_ms = */ 6,
        /* sda_setup_time_ns = */ 227,  /* 227 nsec according to the datasheet */
        /* sck_pulse_width_ns = */ 255, /* 450 nsec according to the datasheet */
        /* sck_clock_hz = */ 1000000UL, /* since Arduino 1.6.0, the SPI bus speed in Hz. Should be  1000000000/sck_pulse_width_ns */
        /* spi_mode = */ 0,             /* active low, rising edge */
        /* i2c_bus_clock_100kHz = */ 4,
        /* data_setup_time_ns = */ 100,
        /* write_pulse_width_ns = */ 100,
        /* tile_width = */ 18,
        /* tile_hight = */ 21,
        /* default_x_offset = */ 0,
        /* flipmode_x_offset = */ 0,
        /* pixel_width = */ 144,
        /* pixel_height = */ 168};

uint8_t dh_u8x8_d_ls013b7dh05_144x168(u8x8_t *u8x8, uint8_t msg, U8X8_UNUSED uint8_t arg_int, void *arg_ptr)
{
  switch (msg)
  {
  case U8X8_MSG_DISPLAY_SETUP_MEMORY:
    u8x8_d_helper_display_setup_memory(u8x8, &dh_u8x8_ls013b7dh05_144x168_display_info);
    break;
  default:
    return u8x8_d_ls013b7dh03_128x128(u8x8, msg, arg_int, arg_ptr);
  }
  return 1;
}

class DH_U8G2_LS013B7DH05_144X168_F_4W_HW_SPI : public U8G2
{
public:
  DH_U8G2_LS013B7DH05_144X168_F_4W_HW_SPI(const u8g2_cb_t *rotation, uint8_t cs, uint8_t dc, uint8_t reset = U8X8_PIN_NONE) : U8G2()
  {
    dh_u8g2_Setup_ls013b7dh05_144x168_f(&u8g2, rotation, u8x8_byte_arduino_hw_spi, u8x8_gpio_and_delay_arduino);
    u8x8_SetPin_4Wire_HW_SPI(getU8x8(), cs, dc, reset);
  }

  void dh_u8g2_Setup_ls013b7dh05_144x168_f(u8g2_t *u8g2, const u8g2_cb_t *rotation, u8x8_msg_cb byte_cb, u8x8_msg_cb gpio_and_delay_cb)
  {
    uint8_t tile_buf_height;
    uint8_t *buf;
    u8g2_SetupDisplay(u8g2, dh_u8x8_d_ls013b7dh05_144x168, u8x8_cad_001, byte_cb, gpio_and_delay_cb);
    buf = u8g2_m_18_21_f(&tile_buf_height);
    u8g2_SetupBuffer(u8g2, buf, tile_buf_height, u8g2_ll_hvline_horizontal_right_lsb, rotation);
  }
};

DH_U8G2_LS013B7DH05_144X168_F_4W_HW_SPI u8g2(U8G2_R1, 10, U8X8_PIN_NONE);`

olikraus, 2022-02-03:
I see mode 2 in the current code:
https://github.com/olikraus/u8g2/blob/ef9fa9f8f8aaa7767c7dd959dcf512433ddc2047/csrc/u8x8_d_ls013b7dh03.c#L209

So you mean, this should be 0, right?

davehummel, 2022-02-03:
Apologies, you are correct. It is currently 2 but should be 0.    This likely applies to all other sharp memory displays, though I have not verified.

olikraus, 2022-02-03:
ok, thanks for clarification.

olikraus, 2022-03-04:
done

closed 2022-03-04 by olikraus