olikraus/u8g2 #1779: about the C++ port

leonardowolf, 2022-02-09:
does the functions home, setCursor and print have any port to the C++ version of the library? I'm writing a aplication using the U8G2 with the C++ port on the STM32F103 and looks like only the arduino version have the cursor manipulation background done in order to those 3 functions work... or did i miss something?

olikraus, 2022-02-09:
The U8g2/U8x8 C++ classes are derived from the Arduino print function which in turn is available only in the Arduino environment:
https://github.com/olikraus/u8g2/blob/b3ffad4d836ee955eaf77e30ff92ea64c235707a/cppsrc/U8x8lib.h#L176-L180

leonardowolf, 2022-02-09:
Yeah, diving more in the library files, i saw that. This technic is very useful because it treats the display as if it were a character display, so the cursor management is done by the library itself. Since this is not implemented out of the Arduino env, I have to do it myself. Thank you for your support

olikraus, 2022-02-09:
> Since this is not implemented out of the Arduino env, I have to do it myself. 

My goal had not been the cursor management, but instead the print functionality. Additionally Arduino users are probably already used to this print function.

closed 2022-02-16 by leonardowolf