olikraus/u8g2 #1783: Is it possible to copy generated text from display for use elsewhere?

TyIsI, 2022-02-14:
Is there a way to capture generated content for/on the display to use on another display?

What I'm trying to implement is this:
- Select text from on-screen menu
- Copy the rendered text from the screen
- Map and send the bitmap to another LED screen

Would this be possible?

Thanks!

olikraus, 2022-02-14:
There is the direct buffer API: https://github.com/olikraus/u8g2/wiki/u8g2reference#direct-access-buffer-api
If you use the `_F_` constructor, then the complete display memory can be accessed.

I once created an example here:

https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/DirectAccess/DirectAccess.ino

TyIsI, 2022-02-14:
Awesome, thank you!

I'll have a go at that. I tried the XBM/PBM functions in the mean time, but without too much success. (It renders oddly.)

olikraus, 2022-03-04:
please reopen for further questions...

closed 2022-03-04 by olikraus