olikraus/u8g2 #1784: WiseChip SSD1309 Display Contrast/Crosstalk Fix?

lax4life001, 2022-02-14:
Hello,

I am quite new to coding an Arduino display. I have access to [this WiseChip display](https://www.wisechip.com.tw/en/product-380065/2-70%E2%80%9D-OLED-Display.html) (UG-2864ASWUG01) with a SSD1309 controller. I am using the u8g2 library for drawing graphics.

I used the GraphicsTest.ino file to test the display and have terrible brightness differences based on the lighting rate of each line (# of ON/OFF pixels). Here is a photo, accurate to what the eye sees, of what the display looks like with the solid bitmap. 
[image]
 
Can this be adjusted in the software? Is this a common or known issue? I have not modified any settings from their defaults. This issue is incredibly stark to me, and any possible reduction would be a huge help for the use case of this display. I have access to smaller OLED displays, without my own code on them, that do not show this phenomenon.

Thank you for any help! I can provide any additional information as needed.

olikraus, 2022-02-15:
This is kind of a common problem: I think the current is too high and can not be delivered from the your power source.
A complete row of enabled pixel will draw more energy (which can't be provided so it appears less bright) instead of some single pixels which receive the full current (so they are brighter).

Ideas:
- Improve power supply. USB ports often do not deliver sufficient current over longer period of time.
- Reduce brightness (use setContrast in U8g2)
- Add extra capacitors between GND and power supply. Looking at you PCB I wonder whether C1 and C10 are soldered correctly. Maybe there is an soldering issue which causes this display problem. I personally would heat up the upper solder pad of C1 and C10 a little bit. Maybe also adding a 47uF or 100uF to C6 may improve the situation.

lax4life001, 2022-02-16:
Thank you for the advice! I'll be testing your suggestions and probably will have more questions. This is all very helpful.

olikraus, 2022-03-04:
closing this for now, feel free to reopen

closed 2022-03-04 by olikraus