olikraus/u8g2 #1792: Using the 12864-06D display, and how does flipmode work?

olikraus, 2022-02-22:
### Discussed in https://github.com/olikraus/u8g2/discussions/1791

<div type='discussions-op-text'>

<sup>Originally posted by **RvanDeelen** February 22, 2022</sup>
So a while ago i bought some displays of aliexpress. As one does. They were explicitly specified as being "right to left written". 
It took me ages to get t the right constructor but in the end i found out that  "U8G2_ST7565_NHD_C12864_1_4W_SW_SPI" will work. Because before this i used U8G2_ST7565_KS0713_1_4W_SW_SPI constructor, but altered the ADC reversal line manually in the library.
So now everything is oriented correctly (writing left to right and no mirrored letters), For my use case i did have to use rotation R2 but that's besides the point.

Something i don't understand tho is how the flipmode works in the constructor code. how can i know if the flipmode instruction actually works looking at the library?</div>

olikraus, 2022-02-22:
The code for flipmode is only sent to the display if the flipmode command (https://github.com/olikraus/u8g2/wiki/u8g2reference#setflipmode) is used. Ideally setFlipMode(0) should be executed automatically by the constructor. Unfortunately this isn't done (I missed this very early in the development now I am too lazy to fix this).
As a result, the commands for flip mode 0 must be implemented in the constructor startup code and in the flipmode 0 sequence itself.

In your case:
https://github.com/olikraus/u8g2/blob/d9402d5e40ca1224a300c39d9baef4051eefb7b5/csrc/u8x8_d_st7565.c#L804

The flip mode sequence is sent here:
https://github.com/olikraus/u8g2/blob/d9402d5e40ca1224a300c39d9baef4051eefb7b5/csrc/u8x8_d_st7565.c#L819-L830

And the commands are repeated in the normal startup sequence of that display:
https://github.com/olikraus/u8g2/blob/d9402d5e40ca1224a300c39d9baef4051eefb7b5/csrc/u8x8_d_st7565.c#L787-L802

Which is actually defined here:
https://github.com/olikraus/u8g2/blob/d9402d5e40ca1224a300c39d9baef4051eefb7b5/csrc/u8x8_d_st7565.c#L59-L73

But it seems, that the flip mode sequences are wrong. and the code for flip mode 0 and startup do not match.
In startup we have the commands 0xa0 and 0xc0, however flipmode 0 will send the wrong code 0xa1 0xc0.
This means, the display will be ok as long as you do not use the flipmode command, but will destroy the content after first use of flipmode.

Is this the problem you observed?

To fix this, I need to change
https://github.com/olikraus/u8g2/blob/d9402d5e40ca1224a300c39d9baef4051eefb7b5/csrc/u8x8_d_st7565.c#L822
to zflip1 and
https://github.com/olikraus/u8g2/blob/d9402d5e40ca1224a300c39d9baef4051eefb7b5/csrc/u8x8_d_st7565.c#L827
to zflip0

hmm looks like a bug in u8g2...

RvanDeelen, 2022-02-22:
I didn't perceive it as a bug at first as it just wasn't documented which constructor would would with my specific display, If i use the  U8G2_ST7565_KS0713_1_4W_SW_SPI one (without any edits or .setfontdirection or anything) it looks like the image below. If i use the other it looks normal. 
[image]
The working constructor doesn't *need* the flipmode to work. but i still like to understand it.

RvanDeelen, 2022-02-22:
Incidentally, the U8G2_ST7565_NHD_C12864_1_4W_SW_SPI constructor also stops the very first (current last) column of pixels being cut off. One line up from the bottom, the right most character is supposed to be a capital V . The clear border arround all the text is just how the display is made. there are no pixels there. I *think* the little black vertical line on the left is either left over in the buffer, or it's that cut off bit of the V that got put over there.

olikraus, 2022-02-22:
> I didn't perceive it as a bug at first as it just wasn't documented which constructor would would with my specific display

Indeed you need to test which constructor fits best to your needs.
The ST7565 is actually controller for up to 132 x 65 pixel.
However if a display has 128x64 pixel, then there are 4x2 = 8 possibilities for the location of the 128x64 window within the 132x65 frame buffer. If the 128x64 is shifted in such a way, that for example the two leftmost columns of the display controller ar not connected to the LCD, then it appears, that some text is missing like in your image. U8g2 basically does not know how the LCD is connected to the controller. Moreover each display vendor can swap the assignment between pixel columns/rows and pins of the display controller. This are 4 more possibilites, so wie have 8x4 = 32 different possibilites how a 128x64 display can be connected to the ST7565. Then there are different technologies which require different driving voltage so on one display it might be to dark on others too light, Which gives more possibilites for the contrast.

So instead of just numbering the constructors from 1 to 32 i just took the first display name with a specific architecture and named it according to that display name (like NHD_C12864). So it indicates that this was once implemented for the New Haven https://www.newhavendisplay.com/nhdc12864a1zfsrgbfbwht1-p-3896.html display. But of course it might also fit to any other display with the same architecture.

So my suggest is this: Please check all of the current constructors for a 128x64 display:

[ST7565 EA_DOGM128](https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#st7565-ea_dogm128)
[ST7565 LM6063](https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#st7565-lm6063)
[ST7565 64128N](https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#st7565-64128n)
[ST7565 ZOLEN_128X64](https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#st7565-zolen_128x64)
[ST7565 LM6059](https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#st7565-lm6059)
[ST7565 KS0713](https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#st7565-ks0713)
[ST7565 LX12864](https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#st7565-lx12864)
[ST7565 ERC12864](https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#st7565-erc12864)
[ST7565 ERC12864_ALT](https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#st7565-erc12864_alt)
[ST7565 NHD_C12864](https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#st7565-nhd_c12864)
[ST7565 JLX12864](https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#st7565-jlx12864)
[ST7565 NHD_C12832](https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#st7565-nhd_c12832)

Ignore the contrast, just observer whether the text is correct or not.
Contast can be adjusted always with the setContrast command.

RvanDeelen, 2022-02-23:
So all constructers for each display type? or just the ones that would work protocol wise, ie 4W SW spi

olikraus, 2022-02-23:
wow, nice summary. 

The jlx is indeed identical to the nhd constructor:
https://github.com/olikraus/u8g2/blob/d9402d5e40ca1224a300c39d9baef4051eefb7b5/csrc/u8x8_d_st7565.c#L686-L689

The contranst should work also for your display, unless you have undefinde the macro here:
https://github.com/olikraus/u8g2/blob/d9402d5e40ca1224a300c39d9baef4051eefb7b5/csrc/u8x8_d_st7565.c#L167-L173

To test the contranst you could also use the set contranst example, which loops over the contrast.

Another way would be to manually send the contrast value with `SendF("ca", 0x81, contranst_value)`.
Note that contranst_value in SendF command is from 0 to 63, but u8g2 expects 0 to 255.

You said you selected the correct constructor, but in your picture it seems to be shiftet by some pixel. I am confused. Is there an issue or: Which constructor / commands had been used to generate the shift in the picture?

RvanDeelen, 2022-02-23:
> You said you selected the correct constructor, but in your picture it seems to be shiftet by some pixel. I am confused. Is there an issue or: Which constructor / commands had been used to generate the shift in the picture?

What i mean by that is that when i used a particular wrong one, (in this case U8G2_ST7565_KS0713_1_4W_SW_SPI one) that was one of the results i got. 
Right now, when using U8G2_ST7565_NHD_C12864_1_4W_SW_SPI u8g2 (U8G2_R2, 13, 11, 10, 9); as a constructor it looks like this. 
[image]
Nothing is shifted and the contrast seems to be fully black on white text.

olikraus, 2022-02-23:
> Nothing is shifted and the contrast seems to be fully black on white text.

ok looks good

> But does software SPI also use that specific start and end transfer behind the scenes

There should be no difference between SW SPI and HW SPI.

RvanDeelen, 2022-02-23:
If you want to test contrast, i only have one kind of display. So for all we know there is no contrast instruction in this one and the changing contrast is just a result of something else funky.

olikraus, 2022-02-23:
hmmm... I never came across a display, which does not allow contrast setup.
Technically there are two options: The controller generates the LCD voltage or the LCD voltage is provided by an external circuit. Indeed if there is an external circuit then the LCD adjustment is not possible. However the internal voltage generation includes an  amplifier with controllable gain (64 steps). I have never seen this controller with an external circuit, in other words for all known displays with this controler the contrast can be adjusted via software commands.

olikraus, 2022-03-04:
fixed flip sequence

closed 2022-03-04 by olikraus