olikraus/u8g2 #1793: national characters are not displayed after otf to bdf conversion

Radee, 2022-02-24:
I converted the Polish font named "brygada_1918_regular.otf" to the bdf format (using ttf2bdf.exe) and then to txt using the bdfconv.exe under Windows (F: \ bdfconv.exe -f 1 -m "32-255" -b 0 -n brygada_1918_12 -o " F: \ Lilygo_fonts_tools \ brygada_1918_12.c "" F: \ Lilygo_fonts_tools \ Brigade 1918-12-r.bdf "). After that I added the result of the conversion to the u8g2_fonts.c file and to u8g2.h, but, unfortunately, Polish characters are ignored when displayed.
Can I ask for an answer, what else I need to pay attention to?

Thank you in advance for your assistance,
Radek

https://www.prezydent.pl/storage/file/core_files/2021/8/5/29b038010dcffbf63d25d2100f6b04ad/brygada_1918-regular.otf

olikraus, 2022-02-24:
THe option ' -m "32-255"'  needs to cover all polish unicode glyphs. According to a quick review, some glyphs have higher numbers (see here https://en.wikipedia.org/wiki/Polish_orthography#Computer_encoding), so probably   ' -m "32-512"' should cover them all.

Another point is this: bdfconv can only convert what is there. If the glyph is not included in the of file or not converted by ttf2bdf then it is not there (ttf2bdf is unknown to me).

Radee, 2022-02-25:
The problem was to use ttf2bdf instead of otf2bdf. otf2bdf generates the entire range of characters, while ttf2bdf only generates up to 255. Polish characters are in the range 32-382. Everything works now, thank you for your help!

closed 2022-02-25 by Radee