olikraus/u8g2 #1798: How to use Icon with text string

JOSSEP-ENG, 2022-02-27:
Hi, olikraus.

There were some problems, so I wrote here.

I want to express the text string below Icon.

u8g2_FirstPage (&u8g2);
do{
u8g2_SetFont(&u8g2,u8g2_font_open_iconic_all_4x_t);            
u8g2_DrawGlyph(&u8g2,10,40,123);
} while(u8g2_NextPage (&u8g2));

u8g2_SetFont(&u8g2, u8g2_font_7x13B_tf);
u8g2_DrawStr(&u8g2,10,60,"TIME");
u8g2_SendBuffer(&u8g2);

but it doesn't work, is there any solution?

[image]
[image]
[image]

This is the screen I want to make.
The screen changes every time you press the button.

please, I want to get help on how to write the code to express the screen above.

olikraus, 2022-02-27:
Do not mix FirstPage/NextPage with SendBuffer

Use this:
```
u8g2_FirstPage (&u8g2);
do{
u8g2_SetFont(&u8g2,u8g2_font_open_iconic_all_4x_t);
u8g2_DrawGlyph(&u8g2,10,40,123);
u8g2_SetFont(&u8g2, u8g2_font_7x13B_tf);
u8g2_DrawStr(&u8g2,10,60,"TIME");
} while(u8g2_NextPage (&u8g2));
```

olikraus, 2022-03-04:
Can we close this issue?

JOSSEP-ENG, 2022-03-04:
Thank you for replying.
I solved this problem.
Have a good day!

closed 2022-03-04 by olikraus