olikraus/u8g2 #1800: ArduinoDroid IDE - does not have sys/types.h file

olikraus, 2022-03-01:
### Discussed in https://github.com/olikraus/u8g2/discussions/1799

<div type='discussions-op-text'>

<sup>Originally posted by **DrDan666** March  1, 2022</sup>
Every example fails to compile… missing sys/types.h file. Here is one compile result...

/sdk/hardware/tools/avr/bin/avr-gcc -no-canonical-prefixes -c -g -Os -w -mmcu=atmega328p -DF_CPU=16000000L -DUSB_VID=null -DUSB_PID=null -DARDUINO=10812 -DIDE="ArduinoDroid" -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO /U8g2/src/clib/mui_u8g2.c -o /librariesBuild/U8g2/clib_mui_u8g2.c.o In file included from /U8g2/src/clib/mui_u8g2.c:95:0:
/U8g2/src/clib/mui.h:45:23: fatal error: sys/types.h: No such file or directory
 #include <sys/types.h>

I can run the 128x64 mono I2C using other libraries. Sure hope you can help. </div>

olikraus, 2022-03-01:
remove "include <sys/types.h>" from mui.h

DrDan666, 2022-03-01:
Works great. Sorry fast fix. 😁

closed 2022-03-02 by olikraus