olikraus/u8g2 #1801: st7565 12864 Problems encountered

RichardsATcn, 2022-03-02:
I have an LCD screen that drives the st7565.
[lx12864D规格书.pdf](https://github.com/olikraus/u8g2/files/8170254/lx12864D.pdf)
This is the manual given by the manufacturer, which contains a C51 program source code.
I transplanted it to Arduino and successfully displayed it. However, when I use u8g2 lib, the type of st7565 cannot be displayed.

The  is the successful code：
`  WriteCommand(0xe2);
  WriteCommand(0xAE); 
  WriteCommand(0xA2); 
 
  WriteCommand(0xA0); 
  WriteCommand(0xC0); 
 
  WriteCommand(0x24); 
 
 WriteCommand(0x81); //Sets V0 
 WriteCommand(0x30); //  00-3f
 
 WriteCommand(0x2F); //voltage follower ON regulator ON booster ON  
 WriteCommand(0xA6); //Normal Display (not reverse display)   0xA7 reverse display 
 WriteCommand(0xA4); //Entire Display Disable   All Pixel ON 
 WriteCommand(0x40); //Set Display Start Line = com0  
 WriteCommand(0xB0); //Set Page Address = 0  
 WriteCommand(0x10); //Set Column Address 4 higher bits = 0 
 WriteCommand(0x00); //Set Column Address 4 lower bits = 1 , from IC SEG1 -> SEG128

 LcmClear(0); 
 WriteCommand(0xAF); //Display ON `
[image]

I tried to modify u8x8_ d_ st7565_ lx12864_ init_ seq
[image]
But there is still no display
[image]

olikraus, 2022-03-02:
My suggestion is to use the LX12864 constructor but set the display contrast to 150 or so (u8g2.setContrast(150))
If you are not able to make all pixel set (with u8g2.setContrast(255)) then your wiring is wrong and needs to be fixed.
If  the display is not correct, try one of the other constructors:

[ST7565 EA_DOGM128](https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#st7565-ea_dogm128)
[ST7565 LM6063](https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#st7565-lm6063)
[ST7565 64128N](https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#st7565-64128n)
[ST7565 ZOLEN_128X64](https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#st7565-zolen_128x64)
[ST7565 LM6059](https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#st7565-lm6059)
[ST7565 KS0713](https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#st7565-ks0713)
[ST7565 LX12864](https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#st7565-lx12864)
[ST7565 ERC12864](https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#st7565-erc12864)
[ST7565 ERC12864_ALT](https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#st7565-erc12864_alt)
[ST7565 NHD_C12864](https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#st7565-nhd_c12864)
[ST7565 JLX12864](https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#st7565-jlx12864)
[ST7565 NHD_C12832](https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#st7565-nhd_c12832)

RichardsATcn, 2022-03-03:
I reconfirmed the wiring and tried these constructors. Still can't get the display. with u8g2. Setcontrast (255) no pixels appear black.

RichardsATcn, 2022-03-03:
Thank you very much for your advice. I try to learn the u8g2 library and try to add this code
[testlcd.ino.txt](https://github.com/olikraus/u8g2/files/8174336/testlcd.ino.txt)

olikraus, 2022-03-03:
> Setcontrast (255) no pixels appear black.

I still assume a wiring issue here. Or maybe you have assigned the wrong pin numbers. Which interface did you use? I suggest to start with SW SPI

RichardsATcn, 2022-03-12:
> > Setcontrast (255) no pixels appear black.
> 
> I still assume a wiring issue here. Or maybe you have assigned the wrong pin numbers. Which interface did you use? I suggest to start with SW SPI

Yeah，I used SW SPI.  TestCode and u8g2 wire define use same code

olikraus, 2022-04-03:
Not sure how to help here, can we close the issue?

closed 2022-04-10 by olikraus