olikraus/u8g2 #1802: Warnings during compilation

ratio-x, 2022-03-02:
Using u8g2 2.31.2 in Arduino IDE 2.0.0-rc3 (Windows 11) library is compiled without errors, but with following warnings:

```
c:\path_to\Arduino\libraries\U8g2\src\clib\u8g2_font.c: In function 'u8g2_GetStrX':
c:\path_to\Arduino\libraries\U8g2\src\clib\u8g2_font.c:1191:10: warning: 'ox' may be used uninitialized in this function [-Wmaybe-uninitialized]
   return ox;
          ^~
c:\path_to\Arduino\libraries\U8g2\src\clib\u8x8_d_s1d15721.c: In function 'u8x8_d_s1d15721_common':
c:\path_to\Arduino\libraries\U8g2\src\clib\u8x8_d_s1d15721.c:71:14: warning: variable 'y' set but not used [-Wunused-but-set-variable]
   uint8_t x, y, c;
              ^
c:\path_to\Arduino\libraries\U8g2\src\clib\u8x8_d_ssd1320.c:339:22: warning: 'u8x8_d_ssd1320_cs1_160x132_init_seq' defined but not used [-Wunused-const-variable=]
 static const uint8_t u8x8_d_ssd1320_cs1_160x132_init_seq[] = {
                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Sketch uses 94640 bytes (9%) of program storage space. Maximum is 983040 bytes.
Global variables use 45976 bytes (17%) of dynamic memory, leaving 216168 bytes for local variables. Maximum is 262144 bytes.

--------------------------
Compilation complete.
```

olikraus, 2022-03-02:
Ah, yes... I will check

olikraus, 2022-03-04:
Hmm.. I was not able to reproduce the "return ox" warning. In fact the warning seems to be wrong.
Fixed the other warnings.

closed 2022-03-04 by olikraus