olikraus/u8g2 #1803: u8g2.getMenuEvent();

Odeku, 2022-03-04:
What function is used instead of   u8g2.getMenuEvent   in STM32？

Odeku, 2022-03-04:
How do I know what button I pressed

Odeku, 2022-03-04:
uint8_t u8x8_GetMenuEvent(u8x8_t *u8x8)
{ 
	if(HAL_GPIO_ReadPin(KEY_GPIO_Port, KEY_Pin) == 0)
	{
		return U8X8_MSG_GPIO_MENU_PREV;			
	}
	while(HAL_GPIO_ReadPin(KEY_GPIO_Port, KEY_Pin) == 0);
	
	if(HAL_GPIO_ReadPin(KEY1_GPIO_Port,KEY1_Pin) == 0)
	{
		return U8X8_MSG_GPIO_MENU_NEXT;			
	}
	while(HAL_GPIO_ReadPin(KEY1_GPIO_Port,KEY1_Pin) == 0);
}

olikraus, 2022-03-04:
I don't know this code. The official supported code is here:

https://github.com/olikraus/u8g2/blob/33843d4ea87ca23d9d0c937d27119262ae23ece2/cppsrc/U8x8lib.cpp#L164-L172

closed 2022-04-03 by olikraus