olikraus/u8g2 #1806: Support for SSD1312 display

nikitalita, 2022-03-07:
https://www.aliexpress.com/item/1005003749452625.html?spm=a2g0o.order_detail.0.0.13c8f19c48DmeU

[SSD1312 Datasheet](https://www.buydisplay.com/download/ic/SSD1312_Datasheet.pdf)
[image]

The make of this specific module seems to be Zhong JingYuan with a model number of ZJY096I0400WG41, but I can't find the datasheet for this particular model.

Works the same as a ssd1306, except it's mirrored:

[image]

nikitalita, 2022-03-08:
Basically, in order to fix this, you have to set segment remap command to A0, but only if the screen is not inverted.

```
  // https://www.buydisplay.com/download/ic/SSD1312_Datasheet.pdf Pg. 51 Section 2.1.19
	//            normal    inverted
	// normal     A1 C8       A0 C0
	// mirrored   A0 C8       A1 C0
	const u8 nSegRemap        = (m_Rotation == TLCDRotation::Inverted && m_Mirror == TLCDMirror::Normal) || 
															(m_Rotation == TLCDRotation::Normal && m_Mirror == TLCDMirror::Mirrored) ? 0xA0 : 0xA1;	
	const u8 nCOMScanDir      = m_Rotation == TLCDRotation::Inverted ? 0xC0 : 0xC8;
```

olikraus, 2022-03-08:
Which constructor did you use to drive the display in the above picture?

I have also ordered the same display.

nikitalita, 2022-03-09:
U8G2_SSD1306_128X64_NONAME_F_HW_I2C

olikraus, 2022-03-09:
Will the execution of `u8g2.sendF("c", 0xa0)` or `u8g2.sendF("c", 0xc0)` after `u8g2.begin()` help?

nikitalita, 2022-03-09:
Yes, sending it after `begin()` seems to work fine as well.

olikraus, 2022-03-09:
What works better, 0xa0 or 0xc0?

nikitalita, 2022-03-09:
As shown in my code example above, it is dependent on whether or not the SSD1312 display is rotated 180 degrees. 
At normal rotation, the Segment Remap command should be 0xA0 and the COM Scan direction command should be 0xC8, but when inverted you want to send 0xA1 and 0xC0, respectively.

olikraus, 2022-03-10:
Thanks for the feedback. Hmm I wonder whether I should create a constructor here.

nikitalita, 2022-03-16:
I think you might want to, given that I am seeing others get very confused by this behavior and being unsure how to correct it. A constructor would ensure the proper commands get sent no matter what the rotation is.

olikraus, 2022-03-19:
I have created beta 2.32.12 which includes the ssd1312 constructor
Can you test, whether this works as expected?

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

nikitalita, 2022-03-26:
> I have created beta 2.32.12 which includes the ssd1312 constructor Can you test, whether this works as expected?

Unfortunately, it does not; it is still mirrored horizontally, no matter what rotation. 
I still have to send the a0 or c0 commands to get it to display properly. 

Using `U8G2_SSD1312_128X64_NONAME_F_HW_I2C`. 
I made sure that I removed the old u8g2 library completely and installed the beta before use.

olikraus, 2022-03-27:
Thanks for the feedback. Meanwhile I received the SSD1312 and I can confirm the problem.
More strange; `U8G2_SSD1312_128X64_NONAME_1_HW_I2C` works as expected. I will check the problem with `U8G2_SSD1312_128X64_NONAME_F_HW_I2C`

olikraus, 2022-03-27:
It should be fixed now. I have created a new beta release...

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

nikitalita, 2022-03-28:
Yes, that works as expected. Thanks!

closed 2022-04-03 by olikraus