olikraus/u8g2 #1809: Initialization can see last screen that was displayed on OLED before power off

Jaboop8, 2022-03-10:
Is there a way to initialize an OLED and have it power on with a blank screen? Currently when I initialize my OLED I see a flash of the screen that was last displayed before powering off my device. I have tried using the clear buffer prior to the initialization but didn't seem to do the trick.

I can see that the OLED turns on immediately after the WriteEscSeqP in the u8g_com_api.c:
uint8_t u8g_WriteEscSeqP(u8g_t *u8g, u8g_dev_t *dev, const uint8_t *esc_seq)

but was unsure if it is possible to send data to the OLED to clear its internal buffer so the screen in blank when after the WriteEscSeq. I am using the old version of this library u8g which I know is not actively maintained so I am just looking for if this is possible and if there is something in the newer library that I can reference that might already be handling this.

Also I have not been able to transition to u8g2 as I need to have the distinction of at least 4 brightness levels on my OLED screen. As far as I have been informed the new library only supports On/Off for pixels am I mistaken?

olikraus, 2022-03-11:
> Is there a way to initialize an OLED and have it power on with a blank screen?

This had been a major goal for u8g2. In fact it should be true for all displays in u8g2 that displays startup with a blank screen.
However this is not true for u8glib.

> As far as I have been informed the new library only supports On/Off for pixels am I mistaken?

Correct. U8g2 only supports pure monochrome (black/white) mode.

Initially my goal was to split u8glib into u8g2 monochrome and a u8g2 graylevel lib. However I never started work on the graylevel lib. So, I am afraid to say: There is no solution for your problem: The blank screen bug was solved, but your display mode is not supported.

Jaboop8, 2022-03-22:
@olikraus Thanks for the insight.

Are there any references to the bug fix that I might be able to add into the u8g library myself? Or is the bug fix reliant on the new architecture of u8g2?

olikraus, 2022-04-03:
> Or is the bug fix reliant on the new architecture of u8g2?

As mentioned above: One reason to create u8g2 had been the above mentioned init problem. There might be a solution for u8g, but any fix will be difficult and time consuming. At least I do not have time to work on u8g lib.

closed 2022-04-10 by olikraus