olikraus/u8g2 #1812: SSD1306 96x39 support

ITstreet1, 2022-03-14:
Just as the title says, how to use this lib with this display. You had support for U8GLIB_SSD1306_96X40 in U8glib that worked pretty much well. Is there a way to use it again? Or the install some earlier version? But which one?

olikraus, 2022-03-14:
I have created u8g2 beta 2.32.11, which includes U8G2_SSD1306_96X40 constructor.
Can you test this now constructor? How will it look like?
Please also execute the page buffer "FlipMode" example. It should show a solid frame all the time. 

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

olikraus, 2022-04-03:
no further feedback, closing

closed 2022-04-03 by olikraus