olikraus/u8g2 #1814: Support screen without set cursor function

xcqq, 2022-03-15:
Hi!
It's a very nice project to give a choice to who do not want to use arduino environent.
I have some questions while porting a new epaper support (I know that this is not recommended, but it's the best mono GUI library I know). The epaper is using UC8151D controller and seems do not support set frame ram cursor before write to frame ram. That means I need to write the whole frame data at one time.
For my understanding, the callback of U8X8_MSG_DISPLAY_DRAW_TILE need support writing data to any coordinate. Is that necessary in current framework?
Thanks

olikraus, 2022-03-15:
U8g2 requires to set the row position. 
For your specific controller, I assume the partial window setup can be used instead of a cursor.
I did something similar for the ssd1606:
https://github.com/olikraus/u8g2/blob/63c719b95ddd19444bea18fafec57d12f3df1f9a/csrc/u8x8_d_ssd1606_172x72.c#L231-L261

xcqq, 2022-03-16:
Great thanks for your advice! I finally got my epaper work now.
BTW, would you like to accept PR to add new epaper support?

closed 2022-04-03 by olikraus